gaifeng8864/klipper-on-android #1: Klipper not installed

d4rk50ul1, 2022-11-28:
Hi @gaifeng8864 , just finished installation following your approach but didn't work. when i run the ending script, after it finishes and i restart the phone, it says: Service 'klipper' not installed. 
here's a couple of outputs from moonraker.log where it shows that apparently klipper service is not running
[image]
[image]
any help would be highly appreciated

gaifeng8864, 2022-11-29:
Have you tried reinstalling the klipper package and running the script to configure the service again? At the same time, you can also run "ps -aux" to check whether services such as klipper are running.

d4rk50ul1, 2022-11-29:
@gaifeng8864 i installed everything all over again changing the user from print3d to print3D... seems like it worked, however, now the issue shown is the following:
[image]
before that one, it was complaining about not finding macros.cfg. fortunatelly, i had a copy of it before.

d4rk50ul1, 2022-11-29:
> @gaifeng8864 i installed everything all over again changing the user from print3d to print3D... seems like it worked, however, now the issue shown is the following: [image] before that one, it was complaining about not finding macros.cfg. fortunatelly, i had a copy of it before.

My Bad, i just realized after backtracking what i did. I added a new printer.cfg setup that i copied from another github. it had both macros.cfg and adxl.cfg files referenced. reason why it is broken.
i'll keep testing and let you know.

gaifeng8864, 2022-11-30:
> My Bad, i just realized after backtracking what i did. I added a new printer.cfg setup that i copied from another github. it had both macros.cfg and adxl.cfg files referenced. reason why it is [...]

I think you are one step away from success, so far the problem is not big.
This error should be that you added an [include adxl.cfg] option in the print.cfg file, but you did not place a real adxl.cfg file in the path /home/print3D/printer_data/config/, So it reports an error because it cannot find this file.
It is recommended that you use the configuration files I provided first (pin definitions need to be modified according to your motherboard model) for testing.
Or you can delete the [include adxl.cfg] option, or place a real adxl.cfg file in the path /home/print3D/printer_data/config/. hope this helps.

d4rk50ul1, 2022-12-06:
Hey @gaifeng8864 
Managed to make it work. You were right,. I was using a printer.cfg pre set up made by other dev. Turns out it considers an adxl accelerometer which was causing errorrs in my setup. So far everything is running smoothly with your config. i'll post more updates if i jump into another issue.

thanks for the help!!

closed 2022-12-06 by d4rk50ul1

gaifeng8864, 2022-12-07:
> Hey @gaifeng8864 Managed to make it work. You were right,. I was using a printer.cfg pre set up made by other dev. Turns out it considers an adxl accelerometer which was causing errorrs in my setup. [...]

This is really good news. Glad to be of help. Thank you for contributing the base script.