gaifeng8864/klipper-on-android #3: Klipper seems connected to smartphone but There is issue with config ig

Saurabh251-cmd, 2022-12-11:
[image]

Saurabh251-cmd, 2022-12-11:
``
[moonraker (1).log](https://github.com/gaifeng8864/klipper-on-android/files/10202030/moonraker.1.log)
look at last

gaifeng8864, 2022-12-11:
You have an 'auto_offset_z' option in your configuration file printer.cfg, which may not be appropriate. This may have something to do with the auto leveling feature.
It is recommended that you refer to the configuration file I provided to adjust the entire process first, so that the entire system can run normally, and then slowly add other functions you need.
The general solution to a problem is to minimize the system and slowly add features until the problem recurs, then you can narrow down the problem and fix it.

Saurabh251-cmd, 2022-12-11:
ohh I will use klipper config for Ender 3 v2 and whenever I restart my phone ip gets changed for linux deploy
Im using octoprint for CH341 driver

Saurabh251-cmd, 2022-12-11:
Finally got it working but no screen on android thoo,how to fix that?
there is bluee screen as well as I can't move Z axis
[image]

gaifeng8864, 2022-12-12:
> Btw how to change Language? i mistakenly changed language to something else while installing

Is it inside debian or inside klipperscreen?

Saurabh251-cmd, 2022-12-12:
> > Btw how to change Language? i mistakenly changed language to something else while installing
> 
> Is it inside debian or inside klipperscreen?

inside debian

gaifeng8864, 2022-12-13:
> > > Btw how to change Language? i mistakenly changed language to something else while installing
> > 
> > 
> > Is it inside debian or inside klipperscreen?
> 
> inside debian

The easiest way is to reinstall the debian system. Or you can change the language option to what you want in linuxdeploy, then run the configuration option in the upper right corner of the main menu. Or you can google how to change the language of linux, there are many tutorials on the Internet.

kahupihu, 2022-12-13:
I remember I once set it to Spanish (es) instead of English (en) and I'm pretty sure @gaifeng8864 is correct.

Saurabh251-cmd, 2022-12-13:
> The easiest way is to reinstall the debian system. Or you can change the language option to what you want in linuxdeploy, then run the configuration option in the upper right corner of the main [...]

Thanks changed

gaifeng8864, 2022-12-13:
> > > > > Btw how to change Language? i mistakenly changed language to something else while installing
> > > > 
> > > > 
> > > > Is it inside debian or inside klipperscreen?
> > > 
> > > 
> > > inside debian
> > 
> > 
> > The easiest way is to reinstall the debian system. Or you can change the language option to what you want in linuxdeploy, then run the configuration option in the upper right corner of the main menu. Or you can google how to change the language of linux, there are many tutorials on the Internet.
> 
> Actually i reinstalled debian but now There is no screen on xerver but running fine fluid on web- [image] how to fix this now

You need to reinstall Xserver XSDL, or you can refer to the discussion in this link: https://github.com/pelya/xserver-xsdl/issues/112

gaifeng8864, 2022-12-13:
> actually i was using on android 6 so maybe thats issue ,i just installed 7 . Im kinda sure it will work here because its blue screen now

Great work. It may be caused by version incompatibility.

Saurabh251-cmd, 2022-12-13:
Thanku so much bro ,now its working perfect as supposed to! Btw i wanna point out that we have to use   git clone https://github.com/th33xitus/kiauh.git because official version giving error in klipper screen installation!
and btw My klipper still using 2 cpu while even used kernel app to unlock remaining cpu

gaifeng8864, 2022-12-14:
> 

I am not sure about the specific reason for this situation, probably because moonraker has some API work that is not suitable. It is recommended that you reinstall the entire klipper family after saving the configuration file. Troubleshooting alone is very troublesome!

Saurabh251-cmd, 2022-12-14:
Yes thanku,
BTW can we do something about tcpuart method to be carried out automatically even if we restart phone?

gaifeng8864, 2022-12-14:
> Yes thanku, BTW can we do something about tcpuart method to be carried out automatically even if we restart phone?

It can be executed, but I don't have the relevant equipment at hand to test it. You can try to follow this command to add it to the automatic startup: sudo unchroot am start -n x.org.server/x.org.server.MainActivity >/dev/null 2>&1
This command is used to start XserverXSDL, you can try to find the package name of tcpuart APP, and then replace "x.org.server/x.org.server.MainActivity".

Saurabh251-cmd, 2022-12-14:
aight ,will try

AyrtonRicardo, 2023-06-18:
> Actually i reinstalled debian but now There is no screen on xerver but running fine fluid on web-
> [image]
> how to fix this now

I had the same issue and I fixed simply replacing the keyboard emulation from the app: https://github.com/pelya/xserver-xsdl/issues/112#issuecomment-1596129361