gaifeng8864/klipper-on-android #4: Automation Enhancements

d4rk50ul1, 2022-12-29:
Hey @gaifeng8864 more than an issue its a comment,
i got a PR from someone that suggested the following lines to Skip Termux usage + automatic selection of Serial Printer:

Termux is not really necessary. You can use unchroot.
Also we can have a permanent link for the serial port from octo4a.

sudo unchroot am start -n com.octo4a/.ui.InitialActivity
sleep 10
sudo ln -s `sudo unchroot readlink -f /data/data/com.octo4a/files/serialpipe` /dev/printer

What do you think?

Saurabh251-cmd, 2022-12-30:
wow seems interesting ,Have u tried this?

gaifeng8864, 2023-01-04:
> Hey @gaifeng8864 more than an issue its a comment, i got a PR from someone that suggested the following lines to Skip Termux usage + automatic selection of Serial Printer:
>
> Termux is not really [...]

Does this mean that almost all motherboards can be automatically recognized and connected to the Android klipper system without compiling the kernel driver for a specific motherboard? If this is the case, I think it is a simple and feasible solution once and for all. How about the memory usage of octo4a APP? Is it a big burden on the system?

Kelsch, 2023-01-06:
> Hey @gaifeng8864 more than an issue its a comment, i got a PR from someone that suggested the following lines to Skip Termux usage + automatic selection of Serial Printer:
>
> Termux is not really [...]

How do you implement this? I am pretty new to linux, so I am might be missing something obvious.

gaifeng8864, 2023-01-08:
> How do you implement this? I am pretty new to linux, so I am might be missing something obvious.

Did you mean octo4a? It's just an APP, just install it directly in Android.

Kelsch, 2023-01-08:
> Did you mean octo4a? It's just an APP, just install it directly in Android.

No, not octo4a. Right now I have 2 printers running Klipper using your tutorial and I am using octo4a for both. I mean where do I need to run these commands to have them make it so that I don't have to change my printer.cfg everytime my phone restarts.

gaifeng8864, 2023-01-11:
> No, not octo4a. Right now I have 2 printers running Klipper using your tutorial and I am using octo4a for both. I mean where do I need to run these commands to have them make it so that I don't have [...]

Is the MCU path generated every time the phone is restarted different? If that's the case, it's really not a good problem to solve.

Kelsch, 2023-01-11:
> Is the MCU path generated every time the phone is restarted different? If that's the case, it's really not a good problem to solve.

Yeah the MCU path goes from /dev/pts/0 to /dev/pts/1 or /dev/pts/4 those are what I have seen it go to on my phone, but others could be different. I was thinking that this line

sudo unchroot readlink -f /data/data/com.octo4a/files/serialpipe /dev/printer

would somehow create a link between whatever /data/data/com.octo4a/files/serialpipe reads, because that is how you find the /dev/pts/{number} and make it a variable or something that would be written into /dev/printer but I guess I was mistaken. My bad.

gaifeng8864, 2023-01-11:
> Yeah the MCU path goes from /dev/pts/0 to /dev/pts/1 or /dev/pts/4 those are what I have seen it go to on my phone, but others could be different. I was thinking that this line
>
> sudo unchroot [...]

This link was posted by @d4rk50ul1. I haven't delved into it. I thought he might have a better solution.

d4rk50ul1, 2023-01-11:
So this is what i've done so far:
1. Check on the unchroot sentence to validate if it effectively detects the octo4a bridge:
sudo unchroot readlink -f /data/data/com.octo4a/files/serialpipe
Result: If you have previously opened Octo4A and let it load fully with the printer connected, it will succesfully show the correct /dev/pts/# device.
2. Create Symlink as it was pointed out to me using the following sentence (be mindful that we're doing command substitution so the backticks are mandatory): 
sudo ln -s `sudo unchroot readlink -f /data/data/com.octo4a/files/serialpipe` /dev/printer
Result: success. You now can use it in printer.cfg to reference the printer in the MCU section.

Issues as of now:

- the symlink stays static after rebooting/shutting down so whenever you reboot your android device, octo4a will change the path but the symlink won't. 
- klipper won't be able to use the symlink unless you chmod it everytime.

Approaches: 
-  i added the following lines to /usr/local/bin/xterm which is one of the scripts that @gaifeng8864  created just to see if it works.:
[image]

i'm currently assessing if it works. I'm very junior on bash scripting so if there's any better approach i think it would be highly appreciated by everyone!

d4rk50ul1, 2023-01-11:
UPDATE: did a small validation just in case the symlink doesn't exists
[image]

d4rk50ul1, 2023-01-11:
Edit: didn't work :S so if @gaifeng8864  could help us that would be awesome!

gaifeng8864, 2023-01-12:
> Edit: didn't work :S so if @gaifeng8864 could help us that would be awesome!

Thank you @d4rk50ul1 for trusting me. But, unfortunately, I'm also a novice at bash scripting.
Does your tweaked setup as in the picture you show work? Can /dev/printer be given 777 permissions automatically after reboot?

d4rk50ul1, 2023-01-27:
Hey @gaifeng8864 i got a suggestion on my original version to apply this very simplistic enhancement and works like a charm!
Once Octo4A is installed and confirmed that your printer is detected. go back to linux deploy and on the container settings do the following: 

On MOUNTS section:
Enable: yes
Mount points: press on the "+" button
Source: /data/data/com.octo4a/files ----this is android's installation folder for octo4a
Target: /home/<user defined>/octo4a ---this should match the user you created
/home/<user defined>/octo4a/serialpipe is the serial port you need to use in your printer.cfg
so now on the MCU section of your printer.cfg, just change it with /home/<user defined>/octo4a/serialpipe
chmod 777 /home/<user defined>/octo4a/serialpipe

and voilah

gaifeng8864, 2023-01-27:
> Hey @gaifeng8864 i got a suggestion on my original version to apply this very simplistic enhancement and works like a charm! Once Octo4A is installed and confirmed that your printer is detected. go back to linux deploy and on the container settings do the following:
> 
> On MOUNTS section: Enable: yes Mount points: press on the "+" button Source: /data/data/com.octo4a/files ----this is android's installation folder for octo4a Target: /home//octo4a ---this should match the user you created /home//octo4a/serialpipe is the serial port you need to use in your printer.cfg so now on the MCU section of your printer.cfg, just change it with /home//octo4a/serialpipe chmod 777 /home//octo4a/serialpipe
> 
> and voilah

Is this method equivalent to mounting the virtual serial port generated by octo4a directly inside centos through linuxdeploy? Then call it directly in printer.cfg?
Is /home//octo4a/serialpipe a fixed name generated by octo4a?

d4rk50ul1, 2023-01-27:
question 1:  yes it's equivalent
question 2: it is a symlink generated by octo4a so after you mount its source directory, you can use it to connect to the printer without any workaround. i'm currently using it and works like a charm!

gaifeng8864, 2023-01-28:
> question 1: yes it's equivalent question 2: it is a symlink generated by octo4a so after you mount its source directory, you can use it to connect to the printer without any workaround. i'm currently using it and works like a charm!

You really did a really great job. In this way, almost all motherboards can be easily connected to the Android phone klipper. Do I need to set octo4a to start automatically if I use this method?

Kelsch, 2023-02-01:
> Hey @gaifeng8864 i got a suggestion on my original version to apply this very simplistic enhancement and works like a charm! Once Octo4A is installed and confirmed that your printer is detected. go [...]

This worked great! I have an issue where octoprint wasn't starting on boot, but once I open that up and chmod 777 /home//octo4a/serialpipe it works!

I would still like to get this figured out to, but this is still easier then changing the mcu everytime as well

gaifeng8864, 2023-02-02:
> This worked great! I have an issue where octoprint wasn't starting on boot, but once I open that up and chmod 777 /home//octo4a/serialpipe it works!
>
> I would still like to get this figured out to, [...]

I'm glad to see that you and @d4rk50ul1 have used this solution at the same time, which shows that the feasibility of this solution has been recognized by more people. According to your and @d4rk50ul1's description, what we need to solve is the problem of octo4a booting up and changing /home//octo4a/serialpipe permissions. I figured this could be done automatically by changing a few things in my adapted script. Let's study the following together.

Kelsch, 2023-02-03:
I read here https://unix.stackexchange.com/questions/663399/restore-default-permissions-on-dev-pts that pts is ram based, so they suggest making a udev rule. I don't know anything about udev or if that's even a reasonable way to go, but I was trying it and this is the rule I was using

KERNAL=="pts*", MODE="0666", ACTION=="add", RUN+="sh /home/print3DchangePermissionspts.sh"

I think we could get away with not using the RUN but just a test I was using it. I wasn't able to get this to work. I think it has to do with the kernel. I just don't think it was running when a new device is plugged in and I don't know how to test it.

gaifeng8864, 2023-02-04:
> Octo4A

Do you know how Octo4A can drive so many devices? If we can know how it works, maybe we can find a more streamlined way to drive the printer motherboard than using Octo4A.

QT7901, 2023-02-24:
hope you can help me, now i am getting Klippy: Disconnected, I did what you said but it doesn't seem to work, I have some config files hope you can take a look and let me know where is it crashing :(
[moonraker.txt](https://github.com/gaifeng8864/klipper-on-android/files/10827250/moonraker.txt)
[printer.txt](https://github.com/gaifeng8864/klipper-on-android/files/10827251/printer.txt)
[fluidd.txt](https://github.com/gaifeng8864/klipper-on-android/files/10827252/fluidd.txt)
[image]

gaifeng8864, 2023-02-25:
> hope you can help me, now i am getting Klippy: Disconnected, I did what you said but it doesn't seem to work, I have some config files hope you can take a look and let me know where is it crashing [...]

Try reinstalling the klipper components, make sure the installation is complete. Often due to network reasons, dependent download failures lead to compilation problems. Then re-execute the script configuration_klipper_family

QT7901, 2023-02-25:
> Reference in new is

After reinstalling, I suddenly noticed this error, but it still installed successfully
[image]

gaifeng8864, 2023-02-25:
> > Reference in new is
> 
> After reinstalling, I suddenly noticed this error, but it still installed successfully [image]

Can I enter the web interface now? Are there any previous errors?

QT7901, 2023-02-25:
> > > Reference in new is
> > 
> > 
> > After reinstalling, I suddenly noticed this error, but it still installed successfully [image]
> 
> Can I enter the web interface now? Are there any previous errors?
[image]
when running command:" bash configuration_klipper_family.sh" => the message pops up "please connect phone..." I followed this video:"https://youtu.be/B5gGMIXwDuo" replace it with "dev/pts/15" when using the command ls -al /data/data/com.octo4a/files/serialpipe => command " bash configuration_klipper_family.sh" run 100% => restart the computer, it will display as shown, I don't know how to fix it here

Kelsch, 2023-02-25:
> 

On my last time installing it I have a had to install it using python 2.7 and not python 3. You can try that and see if that helps.

QT7901, 2023-02-25:
> > 
> 
> On my last time installing it I have a had to install it using python 2.7 and not python 3. You can try that and see if that helps.

I'm currently using version 3. let me reinstall all to see if it works

QT7901, 2023-02-26:
> > 
> 
> On my last time installing it I have a had to install it using python 2.7 and not python 3. You can try that and see if that helps.

it seems to have connected to the printer, now it is giving me error like this, what should i do next :(
[image]

Kelsch, 2023-02-26:
> 

I found this forum post. https://forum.3dprintbeginner.com/t/klipper-and-creality-box/736/5

I think your issue is now just a general klipper/moonraker issue.

gaifeng8864, 2023-02-27:
> > > 
> > 
> > 
> > On my last time installing it I have a had to install it using python 2.7 and not python 3. You can try that and see if that helps.
> 
> it seems to have connected to the printer, now it is giving me error like this, what should i do next :( [image]
This is the CPU temperature display setting I made for my mobile phone Mi 2s. It may not be suitable for your phone. First, you find the following fields in printer.cfg and comment them out, then restart klipper. Normally this should solve the problem. If you want to continue to display the temperature of the phone in the future, you can try to modify the number in the field, for example, change it to 7. The temperature display file serial number of each mobile phone is different, so you need to try it yourself.
[temperature_sensor phone]
sensor_path: /sys/class/thermal/thermal_zone11/temp
sensor_type: temperature_host
min_temp: 10
max_temp: 80

QT7901, 2023-02-27:
> I found this forum post. https://forum.3dprintbeginner.com/t/klipper-and-creality-box/736/5
> 
> I think your issue is now just a general klipper/moonraker issue.

it worked, i flashed the .bin file for mks robin e3, i used the mks printer.cfg file for it and got these errors, what should i do next :(
[image]
[moonraker.txt](https://github.com/gaifeng8864/klipper-on-android/files/10838518/moonraker.txt)
[printer.txt](https://github.com/gaifeng8864/klipper-on-android/files/10838519/printer.txt)
[fluidd.txt](https://github.com/gaifeng8864/klipper-on-android/files/10838520/fluidd.txt)

gaifeng8864, 2023-02-27:
> it worked, i flashed the .bin file for mks robin e3, i used the mks printer.cfg file for it and got these errors, what should i do next :( [image] [...]

It looks like some configuration files are not set completely. Please post your printer.cfg too, and we will analyze the following problems together.

Saurabh251-cmd, 2023-02-28:
> it worked, i flashed the .bin file for mks robin e3, i used the mks printer.cfg file for it and got these errors, what should i do next :( [image] [...]

You can now follow basic klipper installation procedures as You have almost done installing!
https://www.klipper3d.org/Installation.html

gaifeng8864, 2023-02-28:
> it worked, i flashed the .bin file for mks robin e3, i used the mks printer.cfg file for it and got these errors, what should i do next :( [image] [...]

You need to add this configuration item in printer.cfg: [include fluidd.cfg]