gaifeng8864/klipper-on-android #5: Linux Deploy not installing Debian architecture

sssawant17, 2023-02-14:
After following your tutorial (here on Github as well as your video), I am running into following issue:
After hitting "Install", the screen immediately displays ">>> deploy" & "<<< deploy" in a matter of seconds.
upon checking the status,  I get the following response in the screenshot:

[image]

Donor phone: Samsung Galaxy S8 (64 bit processor hence using "Arm64 architecture" option.

gaifeng8864, 2023-02-15:
> After following your tutorial (here on Github as well as your video), I am running into following issue: After hitting "Install", the screen immediately displays ">>> deploy" & "<<< deploy" in a [...]

Can you provide the model of your mobile phone and the version of Android system you are using? And, what method does the Android system use to obtain root privileges?
The reason for this phenomenon is likely to be the incompatibility between the kernel of the Android system and the Debian system or the container virtualized by linuxdeploy. If this is the case, you can only try to change the Android system version.

sssawant17, 2023-02-15:
The issue is resolved by re rooting teh phone with LineageOS

closed 2023-02-15 by sssawant17

gaifeng8864, 2023-02-16:
> The issue is resolved by re rooting teh phone with LineageOS

That's great, congratulations.