gaifeng8864/klipper-on-android #6: unable to use internet as user

PONYMODZ, 2023-03-01:
after following the tutorial I cannot seem to have internet access as my user and this gets me stuck on step 5.
when i try to run
`$ git clone https://github.com/th33xitus/kiauh.git`
i get the result of
```
Cloning into 'kiauh'...
fatal: unable to access 'https://github.com/th33xitus/kiauh.git/': Could not resolve host: github.com
```
if run as root it works but then kiuah installation script complains about root
```
/=======================================================\
|       !!! THIS SCRIPT MUST NOT RUN AS ROOT !!!        |
\=======================================================/
```

Pescu123, 2023-03-01:
Do you have a custom rom installed without Gapps?

I had the same issue when I installed a custom rom without installing the gapps for some reason.

gaifeng8864, 2023-03-02:
> after following the tutorial I cannot seem to have internet access as my user and this gets me stuck on step 5. when i try to run `$ git clone https://github.com/th33xitus/kiauh.git` i get the [...]

sudo usermod -a -G aid_inet,aid_sdcard_rw,aid_net_raw root

I think the above command should solve your problem.

PONYMODZ, 2023-03-02:
> Do you have a custom rom installed without Gapps?
> 
> I had the same issue when I installed a custom rom without installing the gapps for some reason.

reflashed the phone lineageos 19.1 and gaps this time still no success

> sudo usermod -a -G aid_inet,aid_sdcard_rw,aid_net_raw root
> 
> I think the above command should solve your problem.

this didn't help either still same result as before

```
$ git clone https://github.com/th33xitus/kiauh.git
Cloning into 'kiauh'...
fatal: unable to access 'https://github.com/th33xitus/kiauh.git/': Could not resolve host: github.com
```

gaifeng8864, 2023-03-02:
> > Do you have a custom rom installed without Gapps?
> > I had the same issue when I installed a custom rom without installing the gapps for some reason.
> 
> reflashed the phone with lineageos 19.1 and gapps this time, still no success
> 
> > sudo usermod -a -G aid_inet,aid_sdcard_rw,aid_net_raw root
> > I think the above command should solve your problem.
> 
> this didn't help either still same result as before
> 
> ```
> $ git clone https://github.com/th33xitus/kiauh.git
> Cloning into 'kiauh'...
> fatal: unable to access 'https://github.com/th33xitus/kiauh.git/': Could not resolve host: github.com
> ```

sudo usermod -a -G aid_inet,aid_sdcard_rw,aid_net_raw print3D

PONYMODZ, 2023-03-03:
> sudo usermod -a -G aid_inet,aid_sdcard_rw,aid_net_raw print3D

still not working but if i run
```
sudo cat /etc/group
```

it shows that the user has the permission

```
aid_inet:x:3003:print3D,root
aid_net_raw:x:3004:root,print3D
```

PONYMODZ, 2023-03-03:
ok, I found the issue, seems like lineage os 19.1 (android 12L) and upwards broke something within linuxdeploy as it seems.
I run lineage os 17.1 (android 10) without gapps now, I don't know if lineage os 18.1 (android 11) works but it is safe to assume it will not as linuxdeploy targets android 10.

hope this helps some people in the future

closed 2023-03-03 by PONYMODZ

PONYMODZ, 2023-03-03:
this should probably be mentioned in the install guide as a warning for users encountering this issue

gaifeng8864, 2023-03-04:
> this should probably be mentioned in the install guide as a warning for users encountering this issue

Might be a software compatibility issue. After all, the security measures of Android 12 have changed a lot. Operating systems above Android 9 are not recommended.

gaifeng8864, 2023-03-04:
> this should probably be mentioned in the install guide as a warning for users encountering this issue

Special instructions have been added, many thanks.

Special Note: The Android used in the installation solution in this tutorial is best to choose a version between Android 5 and Android 9. Operating system lower than Android 5 need to use a lower version of linuxdeploy, and operating system higher than Android 9 may have permission-related compatibility issues. In addition, if openssh fails to start automatically (shown as unable to use ssh to connect to the debian), this situation is generally a compatibility problem between the Android kernel and linuxdeploy, and you need to replace the Android system with a different kernel version (note that it is a different kernel version).