gaifeng8864/klipper-on-android #7: First use kiauh to install the klipper family!

bobdivx, 2023-03-09:
hello i am trying to install klipper on a huawei p9 lite, root OK.

However, I'm stuck at this step:
$ bash configuration_klipper_family.sh
First use kiauh to install the klipper family!

gaifeng8864, 2023-03-09:
> hello i am trying to install klipper on a huawei p9 lite, root OK.
> 
> However, I'm stuck at this step: $ bash configuration_klipper_family.sh First use kiauh to install the klipper family!

According to the steps of the tutorial, you need to install the klipper family bucket first.

bobdivx, 2023-03-10:
Effectively.  but there was an error and the files in the internal storage got corrupted.  I did a hard reset of the phone, installed everything.  now more than to test

gaifeng8864, 2023-03-11:
> Effectively. but there was an error and the files in the internal storage got corrupted. I did a hard reset of the phone, installed everything. now more than to test

Where are you at now? Can it be used normally?

bobdivx, 2023-03-11:
I'm waiting to receive the printer 😇.  but I can't find a working version of XServer-XSDL.  the latest version does not work