gaifeng8864/klipper-on-android #9: reportsome error in my

cliffmanw, 2023-04-14:


gaifeng8864, 2023-04-14:
> 1. In this option :“安装路径 （/data/debian11）”, I have to change the path to "/data/local/debian", or the ssh will fail at startup.
>    the path is from this link: https://gist.github.com/RyanEwen/ae81fc48ad00397f1026915f0e6beed9#file-readme-md
> 2. must keep the screen always on in android settings, or some weir thing happens on networking, like dns stop working.
> 
> for tty, I think WinTerm is a good free client.

Is it just a network problem? If it's just a network problem, I modified some settings in the script, which should solve the problem of automatic network disconnection.

cliffmanw, 2023-04-14:
> Is it just a network problem? If it's just a network problem, I modified some settings in the script, which should solve the problem of automatic network disconnection.

I doubt it, tested few times, if screen off(power botton pushed),dns fails, pinging www.baidu.com won't work, if screen stay on, it works. every devices on same LAN working good at the same time. tested on rooted huawei P9 EMUI8, android8.0.

and now, still can't get klipper work, checked /dev/, no ttyACM0 listed. trying to fix it by follow the instruction in RyanEwen's link.