gaifeng8864/klipper-on-android #11: Problem with "Linux Deploy"

Photelegy, 2023-06-02:
Hi everyone,

I've just rooted my smartphone (Samsung Galaxy S8+) and now I'm trying to put Klipper on it with this [YouTube-tutorial](https://youtu.be/B5gGMIXwDuo?t=379) which is 
 based on your "README_en.md" from [klipper-on-android](https://github.com/gaifeng8864/klipper-on-android/tree/main).

My problem is:
After setting everything up and changing the settings (see the screenshots) like showed in [2. Install linuxdeploy](https://github.com/gaifeng8864/klipper-on-android/blob/main/README_en.md#2-install-linuxdeploy) and [3. Install debian](https://github.com/gaifeng8864/klipper-on-android/blob/main/README_en.md#3-install-debian) and pressing "Install" the log shows only the following:
```
[17:13:54] >>> deploy
[17:13:54] <<< deploy
```
I still tried what happens when playing "START" after that. Then it shows:
```
[17:13:57] >>> start
[17:13:57] >>> Missing parameters: TARGET_PATH
[17:13:57] <<< start
```
What could be the problem?

[image]

And here my settings:
[image]

Thank you all very much!
Kind Regards
Photelegy

gaifeng8864, 2023-06-03:
> Hi everyone,
> 
> I've just rooted my smartphone (Samsung Galaxy S8+), gave the Apps ("XServer XSDL", "Kernel Adiutor" and "Linux Deploy") root-permission and now I'm trying to put Klipper on it with this [YouTube-tutorial](https://youtu.be/B5gGMIXwDuo?t=379) which is based on your "README_en.md" from [klipper-on-android](https://github.com/gaifeng8864/klipper-on-android/tree/main).
> 
> My problem is: After setting everything up and changing the settings (see the screenshots) like showed in [2. Install linuxdeploy](https://github.com/gaifeng8864/klipper-on-android/blob/main/README_en.md#2-install-linuxdeploy) and [3. Install debian](https://github.com/gaifeng8864/klipper-on-android/blob/main/README_en.md#3-install-debian) and pressing "Install" the log shows only the following:
> 
> ```
> [17:13:54] >>> deploy
> [17:13:54] <<< deploy
> ```
> 
> I still tried what happens when playing "START" after that. Then it shows:
> 
> ```
> [17:13:57] >>> start
> [17:13:57] Missing parameters: TARGET_PATH
> [17:13:57] <<< start
> ```
> 
> **What could be the problem?**
> 
> [image]
> 
> And here my settings: [image]
> 
> Thank you all very much! Kind Regards Photelegy

On the surface, the direct reason is that debian was not installed successfully. If all the settings are done strictly according to the tutorial, then this may be a compatibility problem, which is common in many Android versions.
     Or, you can try to click "remove ENV" on the linuxdeploy setting interface first, and then click "update ENV". If it still doesn't work, you can uninstall linuxdeploy and try to install it again, and strictly follow the tutorial steps again.

Photelegy, 2023-06-03:
> On the surface, the direct reason is that debian was not installed successfully. If all the settings are done strictly according to the tutorial, then this may be a compatibility problem, which is [...]

Thank you very much!
I'll try that again.
But if it would be a compatibility problem, what could I do? Would I need to get another Android version on my Samsung Galaxy S8+?

Kind Regards
Photelegy

Photelegy, 2023-06-03:
> 
> > On the surface, the direct reason is that debian was not installed successfully. If all the settings are done strictly according to the tutorial, then this may be a compatibility problem, which is common in many Android versions.
> >      Or, you can try to click "remove ENV" on the linuxdeploy setting interface first, and then click "update ENV". If it still doesn't work, you can uninstall linuxdeploy and try to install it again, and strictly follow the tutorial steps again.
> 
> 
> 
> Thank you very much!
> I did try all your suggestions.
> 1. I tried "remove ENV" and "update ENV" again. -> didn't work
> 2. I uninstalled and reinstalled Linux Deploy (even tried an older apk-version) and tried again exactly like the tutorial says. (multiple times) -> didn't work
> 
> If it would be a compatibility problem, what could I do? Would I need to get another Android version on my Samsung Galaxy S8+ (model ""SM-G955F")?
> 
> Kind Regards
> Photelegy

gaifeng8864, 2023-06-03:
> On the surface, the direct reason is that debian was not installed successfully. If all the settings are done strictly according to the tutorial, then this may be a compatibility problem, which is [...]

If you have tried the above steps and still have no effect, I suggest changing to an Android version directly. You can try to use the AOSP version of Android, which has better compatibility. They are all constant attempts, which can be regarded as a tossing fun, although sometimes it is very distressing when the problem cannot be solved, haha.

AyrtonRicardo, 2023-06-16:
If you don't have actual magisk running busybox behind the scenes, you need to run busybox yourself:

. install BusyBox app, 
. open the app, 
. select "/system/xbin/" as installation path, 
. leave the newest version, 
. install.

Go back to Linux Deploy app, update your settings "PATH VARIABLE" to "/system/xbin/", click update env, try again.

gaifeng8864, 2023-06-16:
> If you don't have actual magisk running busybox behind the scenes, you need to run busybox yourself:
>
> . install BusyBox app, . open the app, . select "/system/xbin/" as installation path, . leave [...]

It seems that the old version depends on busybox. I have never installed busybox, but it can still run normally.