gaifeng8864/klipper-on-android #12: klipperscreen can't install

maorry, 2023-06-14:
Hello
Trying to install klipperscreen but there's an error
can someone please help

Creating virtual environment
Collecting jinja2==3.1.2 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 1))
  Using cached Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting netifaces==0.11.0 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 2))
  Using cached netifaces-0.11.0.tar.gz (30 kB)
  Preparing metadata (setup.py) ... done
Collecting requests==2.31.0 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 3))
  Using cached requests-2.31.0-py3-none-any.whl (62 kB)
Collecting websocket-client==1.5.3 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 4))
  Using cached websocket_client-1.5.3-py3-none-any.whl (56 kB)
Collecting pycairo==1.23.0 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 5))
  Using cached pycairo-1.23.0-cp37-cp37m-linux_armv7l.whl
Collecting PyGObject==3.44.1 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 6))
  Using cached PyGObject-3.44.1.tar.gz (720 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting python-mpv==0.5.2 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 7))
  Using cached python-mpv-0.5.2.tar.gz (25 kB)
  Preparing metadata (setup.py) ... done
Collecting six==1.16.0 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 8))
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting dbus-python==1.3.2 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 9))
  Using cached dbus-python-1.3.2.tar.gz (605 kB)
  Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [129 lines of output]
      Collecting meson-python>=0.8.1
        Using cached meson_python-0.13.1-py3-none-any.whl (79 kB)
      Collecting meson>=0.60.0
        Using cached meson-1.1.1-py3-none-any.whl (918 kB)
      Collecting ninja
        Using cached ninja-1.11.1.tar.gz (27 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Collecting patchelf
        Using cached patchelf-0.17.2.1-py2.py3-none-manylinux_2_17_armv7l.manylinux2014_armv7l.musllinux_1_1_armv7l.whl (381 kB)
      Collecting setuptools
        Using cached setuptools-67.8.0-py3-none-any.whl (1.1 MB)
      Collecting wheel
        Using cached wheel-0.40.0-py3-none-any.whl (64 kB)
      Collecting pyproject-metadata>=0.7.1 (from meson-python>=0.8.1)
        Using cached pyproject_metadata-0.7.1-py3-none-any.whl (7.4 kB)
      Collecting tomli>=1.0.0 (from meson-python>=0.8.1)
        Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
      Collecting packaging>=19.0 (from pyproject-metadata>=0.7.1->meson-python>=0.8.1)
        Using cached packaging-23.1-py3-none-any.whl (48 kB)
      Building wheels for collected packages: ninja
        Building wheel for ninja (pyproject.toml): started
        Building wheel for ninja (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error
      
        × Building wheel for ninja (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [91 lines of output]
      
      
            --------------------------------------------------------------------------------
            -- Trying 'Ninja' generator
[... 79 lines cut ...]
              Command:
                /usr/bin/cmake /tmp/pip-install-gpyh9d16/ninja_b836fcad66314640836f8a636f2c9849 -G 'Unix Makefiles' --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX:PATH=/tmp/pip-install-gpyh9d16/ninja_b836fcad66314640836f8a636f2c9849/_skbuild/linux-armv7l-3.7/cmake-install -DPYTHON_VERSION_STRING:STRING=3.7.3 -DSKBUILD:INTERNAL=TRUE -DCMAKE_MODULE_PATH:PATH=/tmp/pip-build-env-9ar890gj/overlay/lib/python3.7/site-packages/skbuild/resources/cmake -DPYTHON_EXECUTABLE:PATH=/home/print3D/.KlipperScreen-env/bin/python3 -DPYTHON_INCLUDE_DIR:PATH=/usr/include/python3.7m -DPYTHON_LIBRARY:PATH=/usr/lib/arm-linux-gnueabihf/libpython3.7m.so -DPython_EXECUTABLE:PATH=/home/print3D/.KlipperScreen-env/bin/python3 -DPython_ROOT_DIR:PATH=/home/print3D/.KlipperScreen-env -DPython_FIND_REGISTRY:STRING=NEVER -DPython_INCLUDE_DIR:PATH=/usr/include/python3.7m -DPython3_EXECUTABLE:PATH=/home/print3D/.KlipperScreen-env/bin/python3 -DPython3_ROOT_DIR:PATH=/home/print3D/.KlipperScreen-env -DPython3_FIND_REGISTRY:STRING=NEVER -DPython3_INCLUDE_DIR:PATH=/usr/include/python3.7m -DCMAKE_BUILD_TYPE:STRING=Release
              Source directory:
                /tmp/pip-install-gpyh9d16/ninja_b836fcad66314640836f8a636f2c9849
              Working directory:
                /tmp/pip-install-gpyh9d16/ninja_b836fcad66314640836f8a636f2c9849/_skbuild/linux-armv7l-3.7/cmake-build
            Please see CMake's output for more information.
      
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for ninja
      Failed to build ninja
      ERROR: Could not build wheels for ninja, which is required to install pyproject.toml-based projects
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Error: pip install exited with status code 0

gaifeng8864, 2023-06-14:
The error you're encountering indicates that there was an issue installing the build dependencies for the `ninja` package. This package is required for the installation of `klipperscreen`.

The error message suggests that CMake 3.15 or higher is required, but your current version is 3.13.4. To resolve this issue, you can try upgrading CMake to a newer version. Here's how you can do it:

1. Update your system's package manager to ensure you have the latest package information. Run the following command:

   ```
   sudo apt update
   ```

2. Upgrade CMake using the package manager. Run the following command:

   ```
   sudo apt install cmake
   ```

   This will install or update CMake to the latest version available in the package repositories.

Once the CMake upgrade is complete, you can try installing `klipperscreen` again. If the error persists, please provide the updated error message, and I'll be happy to assist you further.

maorry, 2023-06-14:
Thank you, I tried that but it didn't work for me.
My fix was to delete it all and doing everything from the start, only changing in linuxdiploy Distribution suite (oldstable) instead of (stable) .

Now I have a different problem: 
After installing everything, the installation didn't create /home/print3D/printer_data/config/

-sh: 8: /home/print3D/printer_data/config/: not found

No printer_data directory on my phone

should I open a new issue for it?

gaifeng8864, 2023-06-14:
> Thank you, I tried that but it didn't work for me. My fix was to delete it all and doing everything from the start, only changing in linuxdiploy Distribution suite (oldstable) instead of (stable) . [...]

This path is automatically created when klipper is installed. If not, it means that klipper may not be installed properly. You can try to reinstall klipper, and pay attention to see if there are any errors during the installation process! Also, why are you using "oldstable"?

Dani7710, 2023-06-15:
> The error you're encountering indicates that there was an issue installing the build dependencies for the `ninja` package. This package is required for the installation of `klipperscreen`.
>
> The [...]

hello, I also have problems with cmake that I can't solve,
"sudo apt install cmake" it says that the latest version is already installed, and it's the old one 3.13,

I tried this guide, https://linuxhint.com/install-cmake-on-debian/, but when you try to give the command "cd cmake-3.24.2" it doesn't enter the unzipped folder, and it doesn't give any error. as if the folder doesn't exist. please help me

gaifeng8864, 2023-06-16:
> hello, I also have problems with cmake that I can't solve, "sudo apt install cmake" it says that the latest version is already installed, and it's the old one 3.13,
>
> I tried this guide, [...]

Which version of debian are you using?

Dani7710, 2023-06-21:
> Which version of debian are you using?

i fixed cmake with oldstable and busybox, stable and buster not work for me (same with/whitout busybox) (huawei mediapad t5) now i'm stuck on
mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
if i ssh
cd ~ && ls /dev/serial/by-id/* gives me
ls: cannot access '/dev/serial/by-id/*': No such file or directory
then I installed xubuntu on a notebook and via terminal emulator I received:
/dev/serial/by-id/usb-Klipper_stm32f407xx_420052000A50325436303720-if00
i put it on printer.cfg on mcu, i tested different printer.cfg for my printer (flsun sr) but it doesn't work all the time mcu error
i tested adding -port0, removing if00, same result.
for now when I turn on the tablet, if connected with otg, klipperscreen starts automatically but does not connect to the printer, moonraker works correctly. I just can't understand where is the error because with the notebook it connects using mainsail. help me please

gaifeng8864, 2023-06-22:
> > > > The error you're encountering indicates that there was an issue installing the build dependencies for the `ninja` package. This package is required for the installation of `klipperscreen`.
> > > > The error message suggests that CMake 3.15 or higher is required, but your current version is 3.13.4. To resolve this issue, you can try upgrading CMake to a newer version. Here's how you can do it:
> > > > ```
> > > > 1. Update your system's package manager to ensure you have the latest package information. Run the following command:
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > sudo apt update
> > > > ```
> > > > 
> > > > 2. Upgrade CMake using the package manager. Run the following command:
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > sudo apt install cmake
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > This will install or update CMake to the latest version available in the package repositories.
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > Once the CMake upgrade is complete, you can try installing `klipperscreen` again. If the error persists, please provide the updated error message, and I'll be happy to assist you further.
> > > 
> > > 
> > > hello, I also have problems with cmake that I can't solve, "sudo apt install cmake" it says that the latest version is already installed, and it's the old one 3.13,
> > > I tried this guide, https://linuxhint.com/install-cmake-on-debian/, but when you try to give the command "cd cmake-3.24.2" it doesn't enter the unzipped folder, and it doesn't give any error. as if the folder doesn't exist. please help me
> > 
> > 
> > Which version of debian are you using?
> 
> i fixed cmake with oldstable and busybox, stable and buster not work for me (same with/whitout busybox) (huawei mediapad t5) now i'm stuck on mcu 'mcu': Unable to connect Once the underlying issue is corrected, use the "FIRMWARE_RESTART" command to reset the firmware, reload the config, and restart the host software. Error configuring printer if i ssh cd ~ && ls /dev/serial/by-id/* gives me ls: cannot access '/dev/serial/by-id/*': No such file or directory then I installed xubuntu on a notebook and via terminal emulator I received: /dev/serial/by-id/usb-Klipper_stm32f407xx_420052000A50325436303720-if00 i put it on printer.cfg on mcu, i tested different printer.cfg for my printer (flsun sr) but it doesn't work all the time mcu error i tested adding -port0, removing if00, same result.
> 
> for now when I turn on the tablet, if connected with otg, klipperscreen starts automatically but does not connect to the printer, moonraker works correctly. I just can't understand where is the error because with the notebook it connects using mainsail. i used too sudo chmod 777 /dev/pts/0 otherwise it wouldn't start everything else as per guide. help me please

Try running: ls -al /dev/
Check for a device like ttyACM0. Or you can save the output of running the above command as a TXT file and send it to me.

Dani7710, 2023-06-22:
> ls -al /dev/

> Try running: ls -al /dev/ Check for a device like ttyACM0. Or you can save the output of running the above command as a TXT file and send it to me.

this is what i get, i tested 2 different otg cables and 2 different usb cables. Thank you
$ ls -al /dev/
total 8
drwxr-xr-x. 24 root          root              2560 Jun 22 08:51 .
drwxr-xr-x. 17 root          root              3488 Jun 16 16:52 ..
----------.  1 root          root                 0 Jun 22 08:50 .coldboot_done
drwx--x--x.  2 root          root              1460 Jun 22 08:50 __properties__
crw-rw-r--.  1 aid_system    aid_system    247,   0 Jun 22 08:50 acm
crw-rw-r--.  1 aid_system    aid_radio      10,  91 Jun 22 08:50 alarm
crw-rw----.  1 aid_radio     aid_system     10,  61 Jun 22 08:50 ar
crw-rw-rw-.  1 root          root           10,  92 Jun 22 08:50 ashmem
crw-------.  1 root          root          238,   0 Jun 22 08:50 bastet
crw-rw-rw-.  1 root          root           10,  85 Jun 22 08:50 binder
drwxr-xr-x.  2 root          root                40 Jun 22 08:50 blkio
drwxr-xr-x.  5 root          root              1620 Jun 22 08:50 block
drwxr-xr-x.  3 root          root                60 Jun 22 08:50 bus
crw-------.  1 root          root           10,  63 Jun 22 08:50 cahub
crw-rw----.  1 aid_system    aid_system     10,  58 Jun 22 08:50 chrKmsgPlat
crw-------.  1 root          root            5,   1 Jun 22 08:50 console
crw-rw----.  1 aid_system    aid_system     10,  74 Jun 22 08:50 cpu_dma_latency
drwxr-xr-x.  2 root          root                40 Jun 22 08:50 cpuctl
drwxr-xr-x.  2 root          root                40 Jun 22 08:50 cpuset
crw-------.  1 root          root           10, 236 Jun 22 08:50 device-mapper
crw-rw----.  1 aid_system             1081  10, 254 Jun 22 08:50 dsm
crw-rw----.  1 root          root           10,  67 Jun 22 08:50 dubai
crw-------.  1 root          root          249,   0 Jun 22 08:50 efuse
-rw-r--r--.  1 aid_logd      aid_logd            24 Jun 22 08:50 event-log-tags
drwxrwxr-x.  2 root          root                60 Jun 22 08:50 f2fs-tools
lrwxrwxrwx.  1 root          root                13 Jun 22 08:50 fd -> /proc/self/fd
crw-rw----.  1 aid_system    aid_system     10,  78 Jun 22 08:50 fingerprinthub
crw-rw----.  1 aid_radio     aid_system     10,  62 Jun 22 08:50 flp
drwxr-xr-x.  2 root          root                40 Jun 22 08:50 frz
drwxrwx---.  2 root          aid_system          40 Jun 22 08:51 fscklogs
crw-rw-rw-.  1 root          root            1,   7 Jun 22 08:50 full
crw-------.  1 root          root           10, 229 Jun 22 08:50 fuse
drwxr-xr-x.  2 root          root               100 Jun 22 08:50 graphics
crwxrwx---.  1 root          aid_audio      10,  94 Jun 22 08:50 hifi_misc
cr--------.  1 root          root           10,  80 Jun 22 08:50 hisi_teelog
crw-rw----.  1 aid_system    aid_system     10,  76 Jun 22 08:50 hw_bfm
crw-r-----.  1 root          aid_system     10,  54 Jun 22 08:50 hwbfgdbg
crw-rw-rw-.  1 root          root           10,  84 Jun 22 08:50 hwbinder
crw-rw----.  1 aid_bluetooth aid_bluetooth  10,  57 Jun 22 08:50 hwbt
crw-rw----.  1 aid_system    aid_system     10,  56 Jun 22 08:50 hwfm
crw-rw----.  1 aid_system    aid_system     10,  55 Jun 22 08:50 hwgnss
crw-rw--w-.  1 root          aid_system     10,  87 Jun 22 08:50 hwlog_dubai
crw-rw--w-.  1 root          aid_log        10,  89 Jun 22 08:50 hwlog_exception
crw-rw--w-.  1 root          aid_system     10,  88 Jun 22 08:50 hwlog_jank
crw-rw----.  1 aid_logd      aid_system     10,  77 Jun 22 08:50 hwlog_switch
crw-rw----.  1 aid_system    aid_system     10,  86 Jun 22 08:50 hwlog_tag
crw-rw----.  1 aid_system    aid_camera    239,   0 Jun 22 08:50 imgsysbrg_vdec0
drwxr-xr-x.  2 root          root               160 Jun 22 08:50 input
drwxr-xr-x.  2 root          root                40 Jun 22 08:50 iolimit
crw-rw-r--.  1 aid_system    aid_system     10,  95 Jun 22 08:50 ion
crw-------.  1 root          root           10,  81 Jun 22 08:50 isp_log
crw-------.  1 root          root            1,  11 Jun 22 08:50 kmsg
drwxr-xr-x.  2 root          root                60 Jun 22 08:50 log
crw-------.  1 root          root           10, 237 Jun 22 08:50 loop-control
crw-------.  1 root          root          237,   0 Jun 22 08:50 madapt
crw-rw-rw-.  1 aid_system    aid_system     10,  93 Jun 22 08:50 mali0
crw-rw----.  1 aid_system    aid_camera    252,   0 Jun 22 08:50 media0
crw-rw----.  1 aid_system    aid_camera    252,   1 Jun 22 08:50 media1
crw-rw----.  1 aid_system    aid_camera    252,   2 Jun 22 08:50 media2
crw-rw----.  1 aid_system    aid_camera    252,   3 Jun 22 08:50 media3
crw-rw----.  1 aid_system    aid_camera    252,   4 Jun 22 08:50 media4
drwx------.  3 root          aid_system          60 Jun 22 08:50 memcg
crw-------.  1 root          root           10,  71 Jun 22 08:50 memory_bandwidth
[... 40 lines cut ...]
crw-------.  1 root          root          244,   1 Jun 22 08:50 ttyHwGS1
crw-------.  1 root          root          244,   2 Jun 22 08:50 ttyHwGS2
crw-rw----.  1 aid_system    aid_vpn        10, 200 Jun 22 08:50 tun
crw-rw----.  1 aid_system    aid_bluetooth  10, 239 Jun 22 08:50 uhid
crw-rw----.  1 aid_system    aid_bluetooth  10, 223 Jun 22 08:50 uinput
crw-rw-rw-.  1 root          root            1,   9 Jun 22 08:50 urandom
drwxrwx---.  4 aid_shell     aid_shell           80 Jun 22 08:50 usb-ffs
[... cut ...]
crw-rw-rw-.  1 root          root           10,  83 Jun 22 08:50 vndbinder
crw-rw----.  1 root          aid_audio      10,  82 Jun 22 08:50 voice_proxy_vowifi
crw-------.  1 root          root           10, 130 Jun 22 08:50 watchdog
crw-------.  1 root          root          251,   0 Jun 22 08:50 watchdog0
prw-r-----.  1 root          adm                  0 Jun 22 08:51 xconsole
crw-r--r--.  1 root          root           10,  75 Jun 22 08:50 xt_qtaguid
crw-rw-rw-.  1 root          root            1,   5 Jun 22 08:50 zero

Dani7710, 2023-06-22:
> Try running: ls -al /dev/ Check for a device like ttyACM0. Or you can save the output of running the above command as a TXT file and send it to me.
uploaded file list
https://github.com/Dani7710/my-test/blob/main/file-lists.txt

gaifeng8864, 2023-06-22:
> > Try running: ls -al /dev/ Check for a device like ttyACM0. Or you can save the output of running the above command as a TXT file and send it to me.
> > uploaded file list
> > https://github.com/Dani7710/my-test/blob/main/file-lists.txt

The current information is not easy to judge. You unplug the printer control board from the phone and run the command again to see the difference in the output.

Dani7710, 2023-06-23:
> The current information is not easy to judge. You unplug the printer control board from the phone and run the command again to see the difference in the output.

this is whitout printer connected:
https://github.com/Dani7710/my-test/blob/main/output.txt

gaifeng8864, 2023-06-24:
> this is whitout printer connected: https://github.com/Dani7710/my-test/blob/main/output.txt

Do you see a difference from the two files? I don't seem to notice any changes. If you don't see a difference, then reconnect your printer motherboard to the phone, then restart the phone, and try to execute the following command after entering debian: "dmesg".
Then look for similar keywords in its output file: "Klipper", "tty", "usb", your printer motherboard cpu model, etc., similar keywords. See if there is information that matches your printer motherboard. Or you can just pack the output information and send it to me.

Dani7710, 2023-06-28:
> Do you see a difference from the two files? I don't seem to notice any changes. If you don't see a difference, then reconnect your printer motherboard to the phone, then restart the phone, and try [...]

with the tablet I don't see the usb connection to the printer with dsmeg, now I'm using a galaxy A50, and on dsmeg I found my printer, this:
[ 1871.866771] [0: kworker/0:1:11792] usb 1-1: new full-speed USB device number 4 using xhci-hcd
[ 1872.010547] [0: kworker/0:1:11792] usb 1-1: udev->lpm_capable=0
[ 1872.048615] [0: kworker/0:1:11792] usb 1-1: New USB device found, idVendor=1d50, idProduct=614e
[ 1872.048646] [0: kworker/0:1:11792] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1872.048666][0:kworker/0:1:11792]usb 1-1:Product:stm32f407xx
[ 1872.048684] [0:kworker/0:1:11792] usb 1-1: Manufacturer: Klipper
[ 1872.048702 ] [0:kworker/0:1:11792] usb 1-1: SerialNumber: 420052000A50325436303720

instead using ls -al /dev/ they look the same both connected and disconnected, how can I translate these lines?
the configuration is debian stable, moreover there is another small problem that klipperscreen does not start, neither automatically nor manually, but it is secondary to the connection problem. Thank you

Dani7710, 2023-06-28:
full report

[ 1871.866771]  [0:    kworker/0:1:11792] usb 1-1: new full-speed USB device number 4 using xhci-hcd
[ 1872.010547]  [0:    kworker/0:1:11792] usb 1-1: udev->lpm_capable=0
[ 1872.048615]  [0:    kworker/0:1:11792] usb 1-1: New USB device found, idVendor=1d50, idProduct=614e
[ 1872.048646]  [0:    kworker/0:1:11792] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1872.048666]  [0:    kworker/0:1:11792] usb 1-1: Product: stm32f407xx
[ 1872.048684]  [0:    kworker/0:1:11792] usb 1-1: Manufacturer: Klipper
[ 1872.048702]  [0:    kworker/0:1:11792] usb 1-1: SerialNumber: 420052000A50325436303720
[ 1872.058321]  [0:    kworker/0:1:11792] __alloc_contig_range: 74 callbacks suppressed
[ 1872.058341]  [0:    kworker/0:1:11792] __alloc_contig_range: [8fb38f, 8fb390) PFNs busy
[ 1872.058468]  [0:    kworker/0:1:11792] __alloc_contig_range: [8fb391, 8fb392) PFNs busy
[ 1872.058589]  [0:    kworker/0:1:11792] __alloc_contig_range: [8fb392, 8fb393) PFNs busy
[ 1872.058710]  [0:    kworker/0:1:11792] __alloc_contig_range: [8fb394, 8fb395) PFNs busy
[ 1872.058831]  [0:    kworker/0:1:11792] __alloc_contig_range: [8fb395, 8fb396) PFNs busy
[ 1872.090468]  [0:    kworker/0:1:11792] usb_notify: call_device_notify device
[ 1872.090495]  [0:    kworker/0:1:11792] usb_notify: send_otg_notify event=device_connect(21) enable=1
[ 1872.090511]  [0:    kworker/0:1:11792] usb_notify: extra_notifier_callback event=device_connect(21)
[ 1872.090526]  [0:    kworker/0:1:11792] usb_notify: extra_notify_state+ event=device_connect(21), enable=on
[ 1872.090542]  [0:    kworker/0:1:11792] usb_notify: extra_notify_state- event=device_connect(21), cable=host_id
[ 1872.090556]  [0:    kworker/0:1:11792] usb_notify: check_gamepad_device : product=stm32f407xx
[ 1872.090569]  [0:    kworker/0:1:11792] usb_notify: check_lanhub_device : product=stm32f407xx
[ 1872.090586]  [0:    kworker/0:1:11792] usb_notify: check_device_speed : dev->speed full-speed on
[ 1872.090599]  [0:    kworker/0:1:11792] usb_notify: check_device_speed : o_notify->speed full-speed
[ 1872.090614]  [0:    kworker/0:1:11792] usb_notify: set_hw_param USB device connected - Class : 0x2, speed :

gaifeng8864, 2023-06-29:
> full report
>
> [ 1871.866771] [0: kworker/0:1:11792] usb 1-1: new full-speed USB device number 4 using xhci-hcd [ 1872.010547] [0: kworker/0:1:11792] usb 1-1: udev->lpm_capable=0 [ 1872.048615] [0: [...]

Run the following command to see if there are new discoveries
dmesg | grep 1-1

Send the output information again, and let's analyze it together.

Dani7710, 2023-06-29:
$ dmesg | grep 1-1
[  119.835217]  [0:irq/276-sm5713-: 2494] sm5713_ccic_event_work : usb: DIAES 4-1-1-1-0
[  119.835681]  [0:irq/276-sm5713-: 2494] sm5713_ccic_event_work : usb: DIAES 1-3-1-1-0
[  120.210434]  [2:    kworker/2:2: 5358] usb 1-1: new full-speed USB device number 2 using xhci-hcd
[  120.357446]  [2:    kworker/2:2: 5358] usb 1-1: udev->lpm_capable=0
[  120.395435]  [2:    kworker/2:2: 5358] usb 1-1: New USB device found, idVendor=1d50, idProduct=614e
[  120.395451]  [2:    kworker/2:2: 5358] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  120.395460]  [2:    kworker/2:2: 5358] usb 1-1: Product: stm32f407xx
[  120.395469]  [2:    kworker/2:2: 5358] usb 1-1: Manufacturer: Klipper
[  120.395477]  [2:    kworker/2:2: 5358] usb 1-1: SerialNumber: 420052000A50325436303720
$ 

full dmesg here (somefile) https://github.com/Dani7710/my-test

I think the problem is the robin nano v3 motherboard, I also tried with a galaxy note 3 neo and the result is the same, it doesn't appear on dev, like in the huawei mediapad T5, like in the galaxy a50. I can also try with an old galaxy J3 2016, but I'm afraid it's the same. while as soon as I connect the printer, octoprint recognizes it, but I think it needs some tricks there too (whit old marlin no problem), on the octoprint screen it gives me:
printer is connected
/dev/bus/usb/001/002
stm32f407xx
VID 1d50 / PID 614e
Thank you

Dani7710, 2023-06-29:
this is the printer.cfg of flsun original speederpad which is officially supported and for sale as klipper upgrade, finally sonicpad and speederpad are tablets...
and below mcu has these lines maybe it's useful
[mcu]
#ls /dev/serial/by-id/*
serial: /dev/serial/by-path/platform-5200000.ehci1-controller-usb-0:1.4:1.0
https://flsun3d.com/pages/speeder-pad all configuration inside

gaifeng8864, 2023-07-01:
> $ dmesg | grep 1-1 [ 119.835217] [0:irq/276-sm5713-: 2494] sm5713_ccic_event_work : usb: DIAES 4-1-1-1-0 [ 119.835681] [0:irq/276-sm5713-: 2494] sm5713_ccic_event_work : usb: DIAES 1-3-1-1-0 [ [...]

Take a look at the discussion at the link below to see if it helps you. Especially test this command: ls -l /sys/bus/usb-serial/devices/
https://stackoverflow.com/questions/54630344/is-it-possible-to-do-the-serial-communication-with-dev-bus-usb-bus-device