gaifeng8864/klipper-on-android #13: How are you able to run "sudo"?

AyrtonRicardo, 2023-06-16:
I've tried different solutions to my Note 5 used to control klipper, but I see now way to run "sudo" inside Linux deploy application because of the way it's mounted.

How are you able to run sudo? Is there something different that you have done to your phone besides rooting it?

gaifeng8864, 2023-06-16:
> I've tried different solutions to my Note 5 used to control klipper, but I see now way to run "sudo" inside Linux deploy application because of the way it's mounted.
> 
> How are you able to run sudo? Is there something different that you have done to your phone besides rooting it?

Can't run "sudo" on your device? What is the error?

AyrtonRicardo, 2023-06-17:
> Can't run "sudo" on your device? What is the error?

When I try to run sudo I get the error below when logged in via SSH with "android/print3d" user:
```bash
sudo: PERM_ROOT: setresuid(0, -1, -1): Operation not permitted
```

Inside my root user I can run all root commands, but I can't become root from my normal user :/
Also my phone is rooted.

gaifeng8864, 2023-06-17:
> ```shell
> ```shell
> sudo: PERM_ROOT: setresuid(0, -1, -1): Operation not permitted
> ```
> 
> 
>     
>       
>     
> 
>       
>     
> 
>     
>   
> ```

To enable the user "print3D" to run sudo commands, you can follow these steps:

1. Log in to the system as a user with administrative privileges, such as the root user or a user with sudo access.

2. Open a terminal or command line interface.

3. Run the following command to edit the sudoers configuration file:

   ```
   sudo visudo
   ```

   This command opens the sudoers file using the visudo tool, which performs syntax checks during editing.

4. In the sudoers file, locate the following line:

   ```
   # User privilege specification
   ```

5. Below that line, add the following entry:

   ```
   print3D ALL=(ALL) ALL
   ```

   This line grants the user "print3D" the permission to run any command as any user on all hosts.

6. Save and exit the sudoers file. In visudo, you can use the Ctrl+X shortcut, followed by Y to save the changes.

7. The user "print3D" should now be able to run sudo commands successfully.

Please note that editing the sudoers file is a sensitive operation. Ensure that you grant necessary privileges only to the required users and follow best security practices. If you are unsure or have any concerns about modifying the sudoers file, it is advisable to back up the file or consult with the system administrator or a professional for guidance.

AyrtonRicardo, 2023-06-17:
> The user "print3D" should now be able to run sudo commands successfully

Unfortunately this is not the reason, why user belongs to sudo as well as it's already in the sudoers configuration.

Even creating a completely new user from within the debian, it does not allow me to execute sudo :/

gaifeng8864, 2023-06-18:
> I'm assuming your are saying to login via ssh to the Debian container created, right?
Yes, just use ssh to log into the Debian container.
Log in as root user, then run the following command:
sudo usermod -aG root print3D
If there is an error, please copy the error message to me.

AyrtonRicardo, 2023-06-18:
> If there is an error, please copy the error message to me.

### Reasons for the issue
Gonna write down here the reasons why it was not working for others.

My main setup was: 
- Note 5 N920C rooted
- Stock Samsung ROM (issue is here)
- SELinux in Enforcing mode

With stock rom I was not able to change selinux to permissive, so I had to search around for a ROM with permissive selinux

### How to fix the problem

You need to have permissive selinux or have a policy that allows to execute sudo within linuxdeploy.
I have not tested, but I saw that newer version of magisk has policies which might solve the problem, but unfortunately the newer version after 18.x were not working on my phone :( 
Try to run newer versions of magisk first before changing your rom, but if you're like me using a spare phone, maybe check the next section here to see some cools roms.

### ROM's with permissive selinux which are still active(links are working)

I came accross 2 nice ones: 
- Lineage OS 17.1 running android 10: the guy was really nice and gave me some insights on my problems in his discord. [Link](https://forum.xda-developers.com/t/rom-unofficial-10-0-11-28-2022-lineageos-17-1-s6-s6edge-s6edge-note5.4132641/). Unfortunately not able to run it. Active community really cool to see.
- BlissROM: it has selinux permissive, but also not booting on my phone
- HavocOS: This one worked to me on version 12.11, unfortunately is Android 10, but it's selinux permissive. [Link](https://forum.xda-developers.com/t/rom-discontinued-10-havoc-3-x-n920x-n920t-w8-n9208.3737405/)

I installed the rom, made my phone permissive, so now I don't have the restriction of Knox anymore and I was able to follow your guide without major problems.

If you are reading this and thinking about using this rom: 

### Known problems
- XServer XSDL is not working at the moment in my build, it shows X server couldnt start, change the display number or reboot the phone.
    - So if you really want to have xserver running, these might not be for your, but they made all problems gone for me :D 

Thank you very much for your support and I hope that I was able to explain how I fix it and if someone else is having those problems they can fix it.

btw, samsung stock rom is waaaaaay heavier for this propose, havocos makes my note 5 seems a newer phone lol

gaifeng8864, 2023-06-18:
> btw, samsung stock rom is waaaaaay heavier for this propose, havocos makes my note 5 seems a newer phone lol

First of all congratulations that you finally found the source of the problem and solved it. But I also heard bad news at the same time, Xserver is not working properly. The reason for this problem is that the Xserver is not compatible with the system. Currently, the only solution I know is to replace the operating system. In addition, selinux can be manually closed. But it involves operating in the boot module inside the ROM. It takes some finesse. You can search for related methods.

AyrtonRicardo, 2023-06-18:
> First of all congratulations that you finally found the source of the problem and solved it. But I also heard bad news at the same time, Xserver is not working properly. The reason for this problem [...]

Thank you for supporting on this,

I was able to fix the issue with xserver, updated my previous comment with the link, so no more issues for me.

Once again, thank you!

gaifeng8864, 2023-06-18:
> Thank you for supporting on this,
>
> I was able to fix the issue with xserver, updated my previous comment with the link, so no more issues for me.
>
> Once again, thank you!

This is really excellent news. Is your fix to replace the keyboard with "QWERT"?

AyrtonRicardo, 2023-06-18:
> This is really excellent news. Is your fix to replace the keyboard with "QWERT"?

Basically yes, use QWERTY keyboard instead of system

gaifeng8864, 2023-06-19:
> > This is really excellent news. Is your fix to replace the keyboard with "QWERT"?
> 
> Basically yes, use QWERTY keyboard instead of system

What a fancy solution. How did you find out that this is possible?

AyrtonRicardo, 2023-06-19:
> > > This is really excellent news. Is your fix to replace the keyboard with "QWERT"?
> > 
> > 
> > Basically yes, use QWERTY keyboard instead of system
> 
> What a fancy solution. How did you find out that this is possible?

several people found this information while adb debugging, so I just tried to fix what I saw in their logcat, which was wrong keyboard initiation, so I just change my keyboard type. Initially I had not even seen the keyboard config, but as soon as I saw it I changed to QWERTY, but in general should be changed to anything else that works :D 

One should check logcat to know more about the reason of failure tho.

gaifeng8864, 2023-06-19:
> several people found this information while adb debugging, so I just tried to fix what I saw in their logcat, which was wrong keyboard initiation, so I just change my keyboard type. Initially I had [...]

Yes, your problem-solving ideas are very good. We often blindly look for answers everywhere when we encounter problems, but we don't know that the running log of the program is the first-hand reference material. Thanks again for your sharing, it has been very helpful.

gaifeng8864, 2023-06-19:
Friends, did you choose the "stable" version when you installed Debian in linuxdeploy? If yes, are you on Debian11 (bullseye) or Debian12 (bookworm) after the installation?

AyrtonRicardo, 2023-06-19:
> Friends, did you choose the "stable" version when you installed Debian in linuxdeploy? If yes, are you on Debian11 (bullseye) or Debian12 (bookworm) after the installation?

bullseye was "stable" some time ago, now it should be "oldstable", because the new stable is debian12

gaifeng8864, 2023-06-19:
> > Friends, did you choose the "stable" version when you installed Debian in linuxdeploy? If yes, are you on Debian11 (bullseye) or Debian12 (bookworm) after the installation?
> 
> bullseye was "stable" some time ago, now it should be "oldstable", because the new stable is debian12

Yes, I just found this problem too. On my xiaomi 2S only bullseye works fine. I changed the installation settings section in linuxdeploy to "oldstable".

gaifeng8864, 2023-06-20:
I have two questions here, and I haven't found the answer for a long time. Please help me to see if you have encountered them.
1. Where are our customized UI settings on the web page of fluiddd, and where are their configuration files saved? I just reinstalled klipper due to some upgrade glitches, but fluidd's UI settings are lost (such as dashboard layout, some temperature presets, etc.). Worse, I don't know where their config files are saved and can't back them up. Resetting these is a chore.
2. In the temperature control interface of KlipperScreen, there are 4 preset temperatures, which are marked as different consumable types, but I did not find the option to customize and modify their parameters. You should know that the printing temperature required by each manufacturer's consumables is different. These 4 preset temperature labels cannot customize parameters, which is useless to me.
Have you noticed the above two questions? If you have a solution to the above two problems, please share with me, I would be very grateful.

AyrtonRicardo, 2023-06-20:
> @AyrtonRicardo I have two questions here, and I haven't found the answer for a long time. Please help me to see if you have encountered them.
> 
> 1. Where are our customized UI settings on the web page of fluiddd, and where are their configuration files saved? I just reinstalled klipper due to some upgrade glitches, but fluidd's UI settings are lost (such as dashboard layout, some temperature presets, etc.). Worse, I don't know where their config files are saved and can't back them up. Resetting these is a chore.
> 2. In the temperature control interface of KlipperScreen, there are 4 preset temperatures, which are marked as different consumable types, but I did not find the option to customize and modify their parameters. You should know that the printing temperature required by each manufacturer's consumables is different. These 4 preset temperature labels cannot customize parameters, which is useless to me.
>    Have you noticed the above two questions? If you have a solution to the above two problems, please share with me, I would be very grateful.

Sorry but I do not know because I use mainsail as interface.

I just made a quick search and found this: https://docs.fluidd.xyz/features/presets, hope it's of some help.

gaifeng8864, 2023-06-21:
I read the official documentation of fluidd, although I still haven't found a solution, haha. But thank you very much.

closed 2023-12-10 by AyrtonRicardo