gaifeng8864/klipper-on-android #15: Permission denied when accessing config

jonawald, 2023-08-02:
$ /home/print3D/printer_data/config/
-sh: 8: /home/print3D/printer_data/config/: Permission denied

I worked through the tutorial, but I get stuck at this part. As far as I can find out, there is no /home/print3D directory. At what point does it get made?

gaifeng8864, 2023-08-03:
> $ /home/print3D/printer_data/config/ -sh: 8: /home/print3D/printer_data/config/: Permission denied
> 
> I worked through the tutorial, but I get stuck at this part. As far as I can find out, there is no /home/print3D directory. At what point does it get made?

It is pre-configured when preparing the installation environment in linuxdeploy before installing the debian system.
Check the setting options in linuxdeploy, if necessary, you can take a screenshot.

jonawald, 2023-08-03:
So I reinstalled Linuxdeploy. making sure that all the options were selected as suggested in the tutorial. Now when I try to go to /home/print3D/printer_data/config/
I get: -sh: 6: /home/print3D/printer_data/config/: not found

gaifeng8864, 2023-08-03:
> So I reinstalled Linuxdeploy. making sure that all the options were selected as suggested in the tutorial. Now when I try to go to /home/print3D/printer_data/config/ I get: -sh: 6: /home/print3D/printer_data/config/: not found

At which step does this problem occur? Is debian already installed? Can you take a screenshot?

jonawald, 2023-08-03:
I wish I knew what you wanted a screenshot of. The Linux deploy on my phone? The xterm window with error message? My Linux deploy settings? 

I re-installed Linux on my phone now I need to re-install the kiauh Skripts. 

When I installed Klipper I hit install and walked away. In the morning, it was asking me if I wanted to join the group, but phone was off I could not. Now I can't get back to that Menü

jonawald, 2023-08-03:
So after wiping the phone and starting from scratch while carefully following the directions, I get the same error.

[image]

gaifeng8864, 2023-08-04:
> So after wiping the phone and starting from scratch while carefully following the directions, I get the same error.
> 
> [image]

Looks like a permission issue. Moreover, there is a problem with your command in the screenshot. You just entered a file path without any instructions. You can add sudo before the command to enable root privileges to run the command.