gaifeng8864/klipper-on-android #16: Failed to build msgpack installing klipper extension

nazariiixa, 2023-11-22:
Octo4a works perfectly, but without klipper, please help...

(1/5) Installing can-utils (2021.08.0-r1)
(2/5) Installing linux-headers (5.19.5-r0)
(3/5) Installing py3-cparser (2.21-r0)
(4/5) Installing py3-cffi (1.15.1-r0)
(5/5) Installing py3-greenlet (2.0.1-r0)
Executing busybox-1.35.0-r29.trigger
OK: 402 MiB in 153 packages
Collecting python-can
  Downloading python_can-4.3.0-py3-none-any.whl (262 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 262.1/262.1 kB 1.6 MB/s eta 0:00:00
Collecting msgpack~=1.0.0
  Downloading msgpack-1.0.7.tar.gz (166 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 166.3/166.3 kB 2.3 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: typing-extensions>=3.10.0.0 in /usr/lib/python3.10/site-packages (from python-can) (4.8.0)
Requirement already satisfied: packaging>=23.1 in /usr/lib/python3.10/site-packages (from python-can) (23.2)
Requirement already satisfied: wrapt~=1.10 in /usr/lib/python3.10/site-packages (from python-can) (1.15.0)
Building wheels for collected packages: msgpack
  Building wheel for msgpack (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for msgpack (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [38 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-armv7l-cpython-310
      creating build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/__init__.py -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/exceptions.py -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/ext.py -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/fallback.py -> build/lib.linux-armv7l-cpython-310/msgpack
      running egg_info
      writing msgpack.egg-info/PKG-INFO
      writing dependency_links to msgpack.egg-info/dependency_links.txt
      writing top-level names to msgpack.egg-info/top_level.txt
      reading manifest file 'msgpack.egg-info/SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      warning: no files found matching '*.c' under directory 'msgpack'
      adding license file 'COPYING'
      writing manifest file 'msgpack.egg-info/SOURCES.txt'
      copying msgpack/_cmsgpack.cpp -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/_cmsgpack.pyx -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/_packer.pyx -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/_unpacker.pyx -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/buff_converter.h -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/pack.h -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/pack_template.h -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/sysdep.h -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/unpack.h -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/unpack_define.h -> build/lib.linux-armv7l-cpython-310/msgpack
      copying msgpack/unpack_template.h -> build/lib.linux-armv7l-cpython-310/msgpack
      running build_ext
      building 'msgpack._cmsgpack' extension
      creating build/temp.linux-armv7l-cpython-310
      creating build/temp.linux-armv7l-cpython-310/msgpack
      gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC -I. -I/usr/include/python3.10 -c msgpack/_cmsgpack.cpp -o build/temp.linux-armv7l-cpython-310/msgpack/_cmsgpack.o
      gcc: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
      compilation terminated.
      error: command 'gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for msgpack
Failed to build msgpack
ERROR: Could not build wheels for msgpack, which is required to install pyproject.toml-based projects

gaifeng8864, 2023-11-23:
> Octo4a works perfectly, but without klipper, please help...
>
> (1/5) Installing can-utils (2021.08.0-r1) (2/5) Installing linux-headers (5.19.5-r0) (3/5) Installing py3-cparser (2.21-r0) (4/5) [...]

gaifeng8864, 2023-11-23:
#### Error Analysis and Possible Causes:

According to the provided information, the error mainly occurs during the building of the `msgpack` package. The key information is as follows:

```bash
gcc: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
error: command 'gcc' failed with exit code 1
```

This error indicates that during the building of `msgpack`, the `gcc` compiler is unable to execute `cc1plus`, and the corresponding file or directory cannot be found.

Possible causes and solutions:

1. **Missing C++ Compiler:** The error suggests that `cc1plus` cannot be found, which is often due to the absence of a C++ compiler. Ensure that the correct C++ compiler is installed. On most Linux systems, you can use the following command:

   ```bash
   sudo apt-get install g++
   ```

   If you are using a different Linux distribution, use the respective package manager to install the C++ compiler.

2. **Missing Dependency Libraries:** Another possible reason is the lack of libraries required for the compilation of `msgpack`. Ensure that the relevant development libraries are installed, such as `libmsgpack-dev`:

   ```bash
   sudo apt-get install libmsgpack-dev
   ```

   Alternatively, if your system uses a different package manager, use the corresponding command to install the necessary development libraries.

3. **Update System:** Ensure that your system and packages are up to date to avoid known issues. Use the following commands to update packages:

   ```bash
   sudo apt-get update
   sudo apt-get upgrade
   ```

4. **Check Environment Variables:** Check the `$PATH` environment variable of your system to ensure it includes the correct paths for locating `gcc` and other relevant tools.

After completing the above steps, try installing the package again to see if the issue is resolved.