gaifeng8864/klipper-on-android #17: Printer not showing as tty device/not connecting

wesblake, 2023-12-19:
I have been in the klipper discord trying to figure this out and they pretty much want to push me away from Android and to the Pi, but it seems crazy having a 7 inch tablet collecting dust when it could do this.
So I have followed your guide a few times but get to the same point. Nexus 7 (Google made by Asus) running Cynagenmod rooted based on Android 7, clean install. Every time all the items install fine (Debian, klipper, fluidd, etc) but I think the problem is when I plug in my printer, it shows up in /dev/bus/usb, not as a tty device.
So the only change in /dev is from :
/dev/bus/usb/001
to
/dev/bus/usb/001/001 and /dev/bus/usb/001/002
I installed Octo4a at one point to see what it sees and it detects printer as /dev/bus/usb/001/002 so that's what I've been using in the scripts (configuration_klipper_family.sh and printer.cfg). However, when it finally all boots and goes to connect to printer, it can't connect to mcu.
Oh, I should mention this is a Flsun QQ-S Pro with the newer GD32F303VET6 chip, I have found other forums though on this setup and compiled firmware accordingly so that SHOULD be good.
I am attaching the latest logs, any ideas? I really want to see this work rather than have to buy yet another device and put this thing back in a drawer!
[klippy.log](https://github.com/gaifeng8864/klipper-on-android/files/13718991/klippy.log)
[moonraker.log](https://github.com/gaifeng8864/klipper-on-android/files/13718994/moonraker.log)

eFfeG1, 2024-11-06:
ho lo stesso problema, sei riuscito a risolvere?

gaifeng8864, 2024-11-09:
> ho lo stesso problema, sei riuscito a risolvere?

My answer applies to your situation as well.