gaifeng8864/klipper-on-android #18: Small refinement of the guide

Zetarancio, 2024-02-01:
You can add something related to:
-Disabling lockscreen at boot
-Using IP web Cam to stream the camera feed inside fluid itself

gaifeng8864, 2024-02-01:
> You can add something related to: -Disabling lockscreen at boot -Using IP web Cam to stream the camera feed inside fluid itself

"Disabling lockscreen at boot", is this necessary? Xserver cannot lock the screen after it is started. It can only make the screen black but cannot completely turn off the screen. If you disable locking the screen at startup, the screen will remain constant and the screen life will be shortened.

Zetarancio, 2024-02-01:
You are most definitely right about the screen life. I use to keep it locked and wake the device up when I need it.
Btw thanks for your work. It's easy and amazing.
Do you have any suggestion for which of the many kiauh scripts to use?
I am currently trying to:
-make mainsail work but it seems broken on every install script
-integrate your work into octo4a in order to work around the USB issue that some people seems to have and use the integrated camera since it doesn't work inside android chroot
Do you happen to have any suggestion?

gaifeng8864, 2024-02-02:
> You are most definitely right about the screen life. I use to keep it locked and wake the device up when I need it. Btw thanks for your work. It's easy and amazing. Do you have any suggestion for [...]

I currently use kiauh (https://github.com/dw-0/kiauh), which as of my last install worked fine and is very convenient.I also really want to integrate klipper directly into octo4a, but my skills in this area are limited and I need to learn from scratch, which may not be possible in the short term. As for the camera issue, I am worried that the performance may not be sufficient when using the chroot solution, so I have not explored it in depth. The above two projects are the next step, but they are somewhat difficult.

Zetarancio, 2024-02-04:
Maybe you can consider getting in touch with octo4a developer. It would be amazing to integrate all projectsinto one.
I would say to not bother with camera support in chroot. I tried but was never able to get Crows nest to see my camera device.

gaifeng8864, 2024-02-07:
> Maybe you can consider getting in touch with octo4a developer. It would be amazing to integrate all projectsinto one. I would say to not bother with camera support in chroot. I tried but was never able to get Crows nest to see my camera device.

this is a good idea! Integrating it into one project will be a feat. It not only facilitates the use of enthusiasts, but also makes a great contribution to the recycling and reuse of electronic products and environmental protection!
Are you saying that from your experience, the camera cannot be used in a chroot environment?

Zetarancio, 2024-02-27:
I am sorry I literally miss the point, btw android needs different permission on the app in order to access the camera according to my understanding... I guess that's what octo4a uses