gaifeng8864/klipper-on-android #19: Network permission denied

WHYCRASH, 2024-02-01:
After running,

sudo usermod -a -G aid_inet,aid_net_raw root

Github and kiauh still cannot connect to the ineternet.

Zetarancio, 2024-02-04:
I solved by changing rom on my device.

gaifeng8864, 2024-02-07:
> After running,
> 
> sudo usermod -a -G aid_inet,aid_net_raw root
> 
> Github and kiauh still cannot connect to the ineternet.

I'm very sorry that I missed your message due to my negligence. As the reply below said, replacing the ROM is a quick and easy possible solution.
Or you can reinstall your existing rom, and then run the kiauh script after ensuring that debian in linuxdeploy can connect to the Internet normally.

gaifeng8864, 2024-02-07:
> I solved by changing rom on my device.

It's a good way to solve the problem. Direct troubleshooting takes time and energy and requires certain professional knowledge. Sometimes we can solve the problem in one step by replacing a more compatible ROM. Ha ha

kcajjones, 2024-08-18:
I have the same issue that LinuxDeploy cannot resolve Github.com, yet the device has full internet access.
I've just installed a fresh ROM.

gaifeng8864, 2024-08-19:
> I have the same issue that LinuxDeploy cannot resolve Github.com, yet the device has full internet access. I've just installed a fresh ROM.

Still can't connect to GitHub?

gaifeng8864, 2024-08-20:
> > > I have the same issue that LinuxDeploy cannot resolve Github.com, yet the device has full internet access. I've just installed a fresh ROM.
> > 
> > 
> > Still can't connect to GitHub?
> 
> Correct. The phone will only allow network access with the sudo command but the klipper install script won't run when under sudo.

Run the following command to try:
sudo usermod -a -G aid_inet,aid_net_raw print3D

kcajjones, 2024-08-20:
> Run the following command to try: sudo usermod -a -G aid_inet,aid_net_raw print3D

Unfortunately this makes no difference.

```
$  sudo usermod -a -G aid_inet,aid_net_raw print3d
$   git clone https://github.com/th33xitus/kiauh.git
Cloning into 'kiauh'...
fatal: unable to access 'https://github.com/th33xitus/kiauh.git/': Could not resolve host: github.com

```

gaifeng8864, 2024-08-21:
> Unfortunately this makes no difference.
>
> ```
> $  sudo usermod -a -G aid_inet,aid_net_raw print3d
> $   git clone https://github.com/th33xitus/kiauh.git
> Cloning into 'kiauh'...
> fatal: unable to access [...]

Can you ping other URLs now?

gaifeng8864, 2024-08-22:
> Unfortunately this makes no difference.
>
> ```
> $  sudo usermod -a -G aid_inet,aid_net_raw print3d
> $   git clone https://github.com/th33xitus/kiauh.git
> Cloning into 'kiauh'...
> fatal: unable to access [...]

You can refer to the solution in the following link: https://github.com/dw-0/kiauh/issues/48

lucky13820, 2024-10-06:
> I solved by changing rom on my device.

What rom did you end up using?