gaifeng8864/klipper-on-android #20: Error with sdbus when installing KlipperScreen

kompumaster, 2024-05-27:
Hello!
Try to install it on honor 9 lite. All goes fine, before klipperscreen installation.

How to fix this step?

`...
Installed Misc packages
Creating virtual environment
Ignoring python-mpv: markers 'python_version >= "3.10"' don't match your environment
Ignoring backports.zoneinfo: markers 'python_version < "3.9"' don't match your environment
Collecting jinja2==3.1.4 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 1))
  Using cached jinja2-3.1.4-py3-none-any.whl.metadata (2.6 kB)
Collecting requests==2.32.2 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 2))
  Using cached requests-2.32.2-py3-none-any.whl.metadata (4.6 kB)
Collecting sdbus==0.11.1 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 3))
  Using cached sdbus-0.11.1.tar.gz (74 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  ? Getting requirements to build wheel did not run successfully.
  ¦ exit code: 1
  ?-> [20 lines of output]
      Traceback (most recent call last):
        File "/home/print3D/.KlipperScreen-env/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/print3D/.KlipperScreen-env/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/print3D/.KlipperScreen-env/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-bj6qjxdw/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 325, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
        File "/tmp/pip-build-env-bj6qjxdw/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-bj6qjxdw/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 487, in run_setup
          super().run_setup(setup_script=setup_script)
        File "/tmp/pip-build-env-bj6qjxdw/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 46, in <module>
        File "<string>", line 33, in get_libsystemd_version
        File "/usr/lib/python3.9/subprocess.py", line 528, in run
          raise CalledProcessError(retcode, process.args,
      subprocess.CalledProcessError: Command '('pkg-config', '--modversion', 'libsystemd')' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

? Getting requirements to build wheel did not run successfully.
¦ exit code: 1
?-> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Error: pip install exited with status code 0
Trying again with new tools...

Message from syslogd@localhost at May 27 03:36:46 ...
 kernel:[12538.216125s][pid:28902,cpu4,sudo]check_root: Uid 60000, Gid 60000, try to Privilege Escalate

Message from syslogd@localhost at May 27 03:36:46 ...
 kernel:[12538.223937s][pid:28902,cpu4,sudo]check_root: Uid 60000, Gid 60000, try to Privilege Escalate

Message from syslogd@localhost at May 27 03:36:46 ...
 kernel:[12538.225341s][pid:28903,cpu5,sudo]check_root: Uid 60000, Gid 0, try to Privilege Escalate
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.9).
cmake is already the newest version (3.18.4-2+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Requirement already satisfied: pip in ./.KlipperScreen-env/lib/python3.9/site-packages (24.0)
Requirement already satisfied: setuptools in ./.KlipperScreen-env/lib/python3.9/site-packages (70.0.0)
Ignoring python-mpv: markers 'python_version >= "3.10"' don't match your environment
Ignoring backports.zoneinfo: markers 'python_version < "3.9"' don't match your environment
Collecting jinja2==3.1.4 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 1))
  Using cached jinja2-3.1.4-py3-none-any.whl.metadata (2.6 kB)
Collecting requests==2.32.2 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 2))
  Using cached requests-2.32.2-py3-none-any.whl.metadata (4.6 kB)
Collecting sdbus==0.11.1 (from -r /home/print3D/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 3))
  Using cached sdbus-0.11.1.tar.gz (74 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  ? Getting requirements to build wheel did not run successfully.
  ¦ exit code: 1
  ?-> [20 lines of output]
      Traceback (most recent call last):
        File "/home/print3D/.KlipperScreen-env/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/print3D/.KlipperScreen-env/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/print3D/.KlipperScreen-env/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-tmmxxfjm/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 325, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
        File "/tmp/pip-build-env-tmmxxfjm/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-tmmxxfjm/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 487, in run_setup
          super().run_setup(setup_script=setup_script)
        File "/tmp/pip-build-env-tmmxxfjm/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 46, in <module>
        File "<string>", line 33, in get_libsystemd_version
        File "/usr/lib/python3.9/subprocess.py", line 528, in run
          raise CalledProcessError(retcode, process.args,
      subprocess.CalledProcessError: Command '('pkg-config', '--modversion', 'libsystemd')' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

? Getting requirements to build wheel did not run successfully.
¦ exit code: 1
?-> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Unable to install dependencies, aborting install.

#=======================================================#
 KlipperScreen installation failed!
#=======================================================#

$
`

igo95862, 2024-05-28:
Are you trying to install on ARMv7 target? The issue is that there is no ARMv7 binary wheel for sdbus so it tries to compile it from source. Compiling from source requires the pkg-config to collect the flags that link with libsystemd.

kompumaster, 2024-05-28:
I make this installation following this manual: https://github.com/gaifeng8864/klipper-on-android/blob/main/README_en.md
Guy install it on loder phone, and use same "armhf" debian architecture....

gaifeng8864, 2024-05-29:
The error you encountered while installing KlipperScreen indicates an issue with building the `sdbus` dependency. Specifically, it mentions an inability to find the version of `libsystemd`, likely due to missing development packages. Here are the steps to resolve this issue:

1. **Install `libsystemd-dev`**:
   This package contains the necessary headers and libraries for developing with `libsystemd`. You can install it using the following command:
   ```bash
   sudo apt-get update
   sudo apt-get install libsystemd-dev
   ```

2. **Install `pkg-config`**:
   Ensure that the `pkg-config` tool is installed, as it is used to detect installed libraries on your system. Install it with:
   ```bash
   sudo apt-get install pkg-config
   ```

3. **Retry Installing KlipperScreen**:
   After installing the above dependencies, try installing KlipperScreen again:
   ```bash
   cd /home/print3D/KlipperScreen
   ./scripts/install-klipperscreen.sh
   ```

If you continue to encounter issues, check the following:

- Make sure the Python environment version is 3.9, as the error message mentions some dependencies related to specific Python versions.
- Ensure there are no leftover or incompatible packages in your system that might be causing the installation to fail.

These steps should help resolve the issue you're facing. If the installation still fails, please provide the new error messages for further analysis.

igo95862, 2024-05-29:
To install source package you will also need Python headers `python3-dev`.

kompumaster, 2024-06-01:
Thank you!
Installed 👍

gaifeng8864, 2024-06-02:
> Thank you! Installed 👍

Which component was installed to solve the problem?

kompumaster, 2024-06-02:
I don't know exactly, install all =)

kompumaster, 2024-06-03:
Reinstall and check it - libsystemd-dev