gaifeng8864/klipper-on-android #22: 'mcu': Unable to connect

letiguja, 2024-07-05:
I change the mcu id in the confiduration to  /dev/ttyACM0 but it still gives me the error. I have tried getting the mcu id through kiauh but it gives me an error. with the ls /dev/ command it lists /dev/ttyACM0 i can not figure out the problem

gaifeng8864, 2024-07-06:
The default device path is /dev/ttyACM0.
In addition, perform the following steps to check the user group to which ttyACM0 belongs and grant permissions to the print3D user.

Run the following command to find the serial device path and user group of the printer motherboard identified in the Android phone:

ls -al /dev/
The output information is generally like this:

crw-rw----. 1 aid_radio aid_radio 166, 0 6 29 11:57 ttyACM0
In the above output information, the second aid_radio is the user group of the serial device, and ttyACM0 is the name of the serial device. The complete serial device path is /dev/ttyACM0

Note! ! ! Different printer motherboards may have different names and user groups in different Android phones, please confirm carefully! ! !

After confirming the user group of the device, run the following command to add user print3D to the user group of the serial device, so that user print3D obtains read and write permissions to the serial device.

Here, the previous method of using scripts to obtain serial device permissions is abandoned. Because I found that there is a big delay in using the script to obtain permissions, and there may be problems with "[SDCard updates](https://www.klipper3d.org/SDCard_Updates.html)".

sudo usermod -a -G aid_radio print3D
Continue to execute the following command:

cd ~
###Enter the login user's home directory

sudo wget https://raw.githubusercontent.com/gaifeng8864/klipper-on-android/main/configuration_klipper_family.sh

bash configuration_klipper_family.sh
Note! ! ! If your serial port device path is not the default /dev/ttyACM0

letiguja, 2024-07-06:
> The default device path is /dev/ttyACM0. In addition, perform the following steps to check the user group to which ttyACM0 belongs and grant permissions to the print3D user.
>
> Run the following [...]

hi i have done all of the steps mentioned above when i restart my phone it works like a klipper screen but it says mcu 'mcu': Unable to connect in fluidd.

gaifeng8864, 2024-07-06:
You run ls -al /dev/ttyACM0 and copy the output to me.

letiguja, 2024-07-06:
crw-------. 1 root root 166, 0 Jun  3  1972 /dev/ttyACM0 is the output
i ran sudo usermod -a -G aid_radio print3D, sudo usermod -a -G root print3D and sudo usermod -a -G dialout print3D

gaifeng8864, 2024-07-07:
> crw-------. 1 root root 166, 0 Jun 3 1972 /dev/ttyACM0 is the output i ran sudo usermod -a -G aid_radio print3D, sudo usermod -a -G root print3D and sudo usermod -a -G dialout print3D

It was my oversight, I'm very sorry.
I have updated the latest installation steps and fixed this problem.
Please refer to: https://github.com/gaifeng8864/klipper-on-android/blob/main/README_en.md
You can proceed directly to step 7.

letiguja, 2024-07-07:
thank you so much it worked.

closed 2024-07-07 by letiguja