gaifeng8864/klipper-on-android #25: Pixel 4 上安装后连不到打印机

lucky13820, 2024-10-07:
我的打印机是 Ender 3 v2，用的是 SKR mini e3 v3。在 Pixel 4 上顺利安装成功后，连接到打印机但是看不到串口设备。dmesg | grep klipper 和 dmesg | grep STM32G0B1 没有返回什么结果。对比连接打印机前后的  ls -al /dev/ 输出结果也看不到有什么不一样的设备。不清楚是我哪块儿没设置好，还是我的打印机的问题。不知道该怎么排查好。

gaifeng8864, 2024-10-07:
> 我的打印机是 Ender 3 v2，用的是 SKR mini e3 v3。在 Pixel 4 上顺利安装成功后，连接到打印机但是看不到串口设备。dmesg | grep klipper 和 dmesg | grep STM32G0B1 没有返回什么结果。对比连接打印机前后的 ls -al /dev/ 输出结果也看不到有什么不一样的设备。不清楚是我哪块儿没设置好，还是我的打印机的问题。不知道该怎么排查好。

你的打印机主板的串口可能是通过ch340转接的，你可以检查一下电路板的USB接口部分。或者直接拍照发来看看。

lucky13820, 2024-10-07:
[image]

查了下 reddit，有人提到他们在 Windows 上安装 CH340 的驱动后就可以连接到这块主板了。所以应该就是 CH340 转接的，那这样的话我应该怎么办？

gaifeng8864, 2024-10-07:
> ![IMG_6319](https://private-user-images.githubusercontent.com/7293811/374020264-501e7fd7-7f90-4469-9bd6-8fbc31620227.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgyODE0NDUsIm5iZiI6MTcyODI4MTE0NSwicGF0aCI6Ii83MjkzODExLzM3NDAyMDI2NC01MDFlN2ZkNy03ZjkwLTQ0NjktOWJkNi04ZmJjMzE2MjAyMjcuanBlZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA3VDA2MDU0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA3YzYxMGMzNjNlZTUzYzI0NjUxZWZkMDFkMzhkY2NhOWJhNDVjNzBiOGVjYjgyY2I4MTAwOTBjMWRhYTY1MjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.VrnGqMa4K3pDKYF1QiJcQvW1-vj7uiFJztC2YvF5-V0)
> 
> 查了下 reddit，有人提到他们在 Windows 上安装 CH340 的驱动后就可以连接到这块主板了。所以应该就是 CH340 转接的，那这样的话我应该怎么办？

目前只有一个办法，就是编译内核，然后在编译选项里打开ch340的驱动，然后把编译后的内核替换到安卓系统里。或者直接编译ch340驱动也可以，单独把驱动编译出来，安装到安卓系统里。