gaifeng8864/klipper-on-android #27: Help, TCP - UART problem

HelioNM, 2024-10-29:
I follow the steps and I cant connect to my printer, the usb serial is not present at the deploy, also I cant use the octo4a method. So I tried with the TCP-UART app. At the first try it works!! klipper was runnig and without errors. Then restarted my phone and... Nothing works again.

The problem is... when I put the socat command this is what happen:

$ socat -d -d pty,link=/tmp/tcpserial,raw,echo=0,waitslave tcp:localhost:8080
2024/10/29 19:41:17 socat[16715] N PTY is /dev/pts/7
2024/10/29 19:42:30 socat[16715] N opening connection to AF=2 127.0.0.1:8080
2024/10/29 19:42:30 socat[16715] N successfully connected from local address AF=2 127.0.0.1:45489
2024/10/29 19:42:30 socat[16715] N starting data transfer loop with FDs [5,5] and [6,6]

Also at the app I can see bytes transfered at the USB_UART -> TCP and TCP -> USB_UART

[image]

The problem is... At the moment I enter the mainsail web and try to start the connection with my printer the connection fail...

2024/10/29 19:49:29 socat[16789] N read(5, 0x82813000, 8192): Input/output error (probably PTY closed)
2024/10/29 19:49:29 socat[16789] N socket 1 (fd 5) is at EOF

[image]

Cant understand why it works so fine the first time I tried this method and after a phone reboot it never works again.

Log: 
[klipy.log](https://github.com/user-attachments/files/17561792/klipy.log)

The phone is in SELinux Permissive btw however I cant no see any diference between permissive and enforcing.
Octa4a is active with the service stopped, also tried to uninstall octa4a however same results.

If anyone can help me...

BTW: the printer connects with no issues to octa4a in this smartphone when the firmware is marlin and right now with the klipper firmware it also works with no issue at my computer with debian. Also it worked fine the first time I tried the method with tcp/uart until I rebooted my phone.

HelioNM, 2024-10-29:
Ok, I found a behaviour, If I go to the mainsail page, hit restart, wait and just at that moment, after the timeout error I put in my ssh window at my PC the socat command, THEN it will work! 

This is a problem, if I close the ssh connectios, my PC goes down, the ssh connection pauses due to inactivity, etc... The printer STOP... 

Any idea? how I can tell to debian to execute the socat command after the klipper is loaded with no ssh connection? and what happens if I need to restart the firmware due a change in config? 

I need help xD

Milliw, 2024-11-17:
Have you gotten any further?