gaifeng8864/klipper-on-android #28: Can't finish installing Debian (oldstable/stable)

Milliw, 2024-11-15:
Hi!

I tried installing both versions several times with also trying to hunt errors (depencencies issues I'd describe it) down and manually install packages. Without success.

Looking at the LinuxDeploy repo it is rather old and to me it looks like creating an issue there will likely have no success.
Has anyone of you folks here using it for this purpose come over similar issues and can help?

Because I'm using a quite old tablet for it (Android 4.4) I was not able to use latest version. Instead I use 2.5.1-257.
I'd really like to succeed and give this old device a new life instead of throwing it away some time. And it would be much more convenient at cheaper, too.

Sadly on both OS SSH was not working so I tried my luck with telnet on tablet/from PC with browser. Don't know if that makes a difference though.

gaifeng8864, 2024-11-16:
> Hi!
> 
> I tried installing both versions several times with also trying to hunt errors (depencencies issues I'd describe it) down and manually install packages. Without success.
> 
> Looking at the LinuxDeploy repo it is rather old and to me it looks like creating an issue there will likely have no success. Has anyone of you folks here using it for this purpose come over similar issues and can help?
> 
> Because I'm using a quite old tablet for it (Android 4.4) I was not able to use latest version. Instead I use 2.5.1-257. I'd really like to succeed and give this old device a new life instead of throwing it away some time. And it would be much more convenient at cheaper, too.
> 
> Sadly on both OS SSH was not working so I tried my luck with telnet on tablet/from PC with browser. Don't know if that makes a difference though.
> 
> EDIT: `Failure trying to run: dpkg --force-depends --install /var/cache/apt/archives/base-passwd_3.5.51_armhf.deb See //debootstrap/debootstrap.log for details (possibly the package /var/cache/apt/archives/base-passwd_3.5.51_armhf.deb is at fault)`
> 
> Is what I get on oldstable version. Trying to fix with `linuxdeploy shell apt --fix-broken install` does nothing. `linuxdeploy shell dpkg --force-depends --install /var/cache/apt/archives/base-passwd_3.5.51_armhf.deb` leads to `:: Configuring core/mnt ... :: Configuring core/net ... Unpacking base-passwd (from /var/cache/apt/archives/base-passwd_3.5.51_armhf.deb)... dpkg: preinst failed, exit code 32512` which I can't resolve myself.
> 
> On "stable" version I managed to get the system running somehow with apt fix broken-install and installed ssh afterwards. But can't login to ssh. My guess is due to missing configuration due to not finishing install script.

The problem of ssh service not being able to start automatically and being unable to log in is common.
According to my experience, this has a lot to do with the Android kernel version. Of course, I have only explored to this stage and have only drawn this empirical conclusion.
It is not clear whether there is a universal solution.

Milliw, 2024-11-17:
I succeeded with Debian Buster. Linux System is running now and I could install all Klipper components. Only had to hard set KlipperScreen to older version because of too old Python (3.7 iirc).

But now I can't see any device when connecting to the printer. Tried with a USB-Stick to the tablet and it got recognized in Android but not in Debian as well (with ls -a /dev/). :-/

Milliw, 2024-11-17:
> The problem of ssh service not being able to start automatically and being unable to log in is common. According to my experience, this has a lot to do with the Android kernel version. Of course, I [...]

AFAIK now (by doing research) it is because the installation scripts(?) of linuxdeploy are too old and thus produce errors with "more recent" Debian installations. I also found some repos of "Linux Deploy Pro" which where linked in a issue thread and claim to have more recent scripts that resolve those issues. But I did not feel safe using those instead "og" linuxdeploy.

Milliw, 2024-11-17:
Hm. I installed usbutils package and at least the devices DO get registered in the linux system...

gaifeng8864, 2024-11-19:
> Hm. I installed usbutils package and at least the devices DO get registered in the linux system...

In other words, have you solved both the USB to serial port driver problem and the ssh failure to start automatically problem?

Milliw, 2024-11-19:
Did you see the other posts I did before?

I did not have an autostart SSH issue but the issue that the OS and especially the SSH parts were not installed. After that got fixed it started without issue.

USB regarding I'm not sure. Devices do get recognised with lsusb but nothing more. Tell me if you like more details about that.
-- Von /e/OS E-Mail gesendet.

gaifeng8864, 2024-11-20:
> AFAIK now (by doing research) it is because the installation scripts(?) of linuxdeploy are too old and thus produce errors with "more recent" Debian installations. I also found some repos of "Linux [...]

I would like to know more about your specific solution here. If you are interested, you can share it separately so that we can benefit future generations. Thank you in advance.

gaifeng8864, 2024-11-20:
> AFAIK now (by doing research) it is because the installation scripts(?) of linuxdeploy are too old and thus produce errors with "more recent" Debian installations. I also found some repos of "Linux [...]

I'd really like to know your specific solution. If you're interested, you can share it separately so that we can benefit future DIY players. Thank you in advance.

Milliw, 2024-11-22:
Hey!
Sorry, I've been busy with other things...

As I did not work on this project for couple of days I'm not sure, what I need to tell any more.

Main thing afair is that I installed that old Debian Buster by manually selecting it.
AFAIR that worked quite well except SSH stuff. Which I could resolve by using above mentioned apt-commands to get it all installed.

Then I installed the KIAUH stuff and had to work around issue with old Python version 3.7 due to old Debian version. Found online that you can force that specific component to old version which works with that old Python version.

I succeeded to install all KIAUH components but then the issue with the USB stuff appeared which I could not manage to get around.
Because of that I tried to upgrade to Debian 11 but there were too many issues which I could not resolve and at that point I gave up.

If you want, I can do everything again and stop before upgrading Debian. Should not be much of a deal. But when we are not able to resolve that USB issues there will be not much benefit from it I guess. AFAIU other people have had same/similar issue or at least that's what I've read here in the issues threads. And not sure, too if it all makes sense because even if we/I will get it working, it's on a very old distro, too.

Just let me know.
Otherwise my next step would be trying on an old Xperia phone - but tablet would be nicer and I guess installation/Debian stuff will be the same nevertheless?!