gaifeng8864/klipper-on-android #30: Power fix causing android to think that battery is 98%

markysmid, 2025-03-10:
Hi, 
First I'm sorry to creating this like an issue. But I tried to understand this but I clearly can't by myself. And I don't know how to ask you just with message.

Your script is perfectly working for me, even the ttyfix is perfect. I had a lot of problems with it before I tried your scripts. Thank you for your work! But I am bit worry about android stil showing charging and battery percentage 98%. When I use OTG + charge (my phone supports this), does it mean that charging will not stop ? Or worse can it destroy my battery after some time when android thinks the battery is 98% ?

So I am thinking about: Why is it there, what is it good for ? How is it right to conect it then ? Is otg with charge ok ? 

Thank you mate.

gaifeng8864, 2025-03-14:
> Hi, First I'm sorry to creating this like an issue. But I tried to understand this but I clearly can't by myself. And I don't know how to ask you just with message.
>
> Your script is perfectly working [...]

The problem you mentioned is not a bug in itself, but the result of a command I deliberately added. That command is mainly to solve the problem of false alarm of the Android system power meter when there is no battery. In order to facilitate power supply, many mobile phones are directly connected to the battery protection board after the battery is removed. This causes a problem. The Android system still thinks that it is still powered by the battery, which causes the system's power meter to start working. The most intuitive phenomenon is that although we use a more durable direct power supply, the system light is still decreasing. In order to solve this problem, I added a piece of code to the script to force the power to be maintained at 98%. As for whether it will damage your battery, it varies from model to model. However, in general, we recommend that you remove the battery and use the direct power supply mentioned above, which is more stable and safe.

markysmid, 2025-03-14:
I see, big thanks for the explanation. So if I would like to run phone with the battery anyway I can delete file "/etc/init.d/powerfix" after running your script, and it should not start ? Or is there something else I have to do ? 
(option is to donwload your script and remove all powerfix parts, but I would like to find a easier way to start the powerfix in case I would need it)
In case I have any troubles with it, I will follow your recommendation.

gaifeng8864, 2025-03-22:
> I see, big thanks for the explanation. So if I would like to run phone with the battery anyway I can delete file "/etc/init.d/powerfix" after running your script, and it should not start ? Or is [...]

You can delete the file "/etc/init.d/powerfix" directly after running the original script, and it will no longer execute the 98% power. Or you can download the script to your local computer, edit and delete the commands related to powerfix, save the changes and then execute the script. If you need to enable the powerfix function again in the future, you can just run the original script again.

markysmid, 2025-03-25:
ok thank you

closed 2025-03-25 by markysmid